BelfrySCAD/BOSL2 #189: Should functions accept $closed to indicate closed paths?

adrianVmariano, 2020-06-23:
It might simplify code if we used $closed as an option for specifying that paths are closed.  This would allow things like:

    let($closed=true) stroke(smooth_path(offset(path_tangent(path))))

rather than having to specify closed=true on every single call.

revarbat, 2020-07-11:
To be honest, this feels like something that will come back to bite us if implemented.

adrianVmariano, 2020-07-11:
So you're saying you're nebulously concerned, but not sure why?

revarbat, 2020-07-11:
This is relying on the bizarre side effects of $specialvars in a way that will hide bugs where calls inside nested functions get a $closed value that is not apparent at its call level.

revarbat, 2020-07-11:
ie: the $closed variable can affect calls that are INSIDE the functions that the $closed is meant to be passed to.

revarbat, 2020-07-11:
```openscad
function pathlen(x) = path_length(x);
...
let($closed=true) stroke(smooth_path(offset(path,r=pathlen(openpath))));
```

revarbat, 2020-07-11:
I'm no longer nebulously concerned.  Not I'm concretely concerned.

adrianVmariano, 2020-07-11:
I do see the point.  This could be addressed by doing

    function pathlen(x) = path_length(x,closed=false);

but yeah, in general I think you're right that it's asking for trouble.

revarbat, 2020-07-11:
I'll mark this one as "will not implement", then.

closed 2020-07-11 by revarbat