BelfrySCAD/BOSL2 #190: Please adapt examples for skin() (and possibly for other modules) to make them fully functional

nesdnuma, 2020-07-01:
Hello,

I have been trying to run the examples for skin(). However , there is code missing in order to make this code functional.

For instance, the first example is: `skin([octagon(4), circle($fn=70,r=2)], z=[0,3], slices=10);`

It would be easier, especially for beginners in OpenSCAD if the code included all the calls to the required files, like this:

'use <BOSL2/skin.scad>
use <BOSL2/std.scad>

skin([octagon(4), circle($fn=70,r=2)], z=[0,3], slices=10);'

Thank you.

adrianVmariano, 2020-08-02:
What is the thinking on this request?

revarbat, 2020-08-02:
Files not part of std.scad don't include the include lines, and that's confusing for new users, apparently.

adrianVmariano, 2021-01-05:
Includes are now included in the examples.

closed 2021-01-05 by adrianVmariano