BelfrySCAD/BOSL2 #200: Q_Slerp

RonaldoCMP, 2020-07-08:
I have compared the outputs of Q_Slerp with the output of an implementation of a Slerp code found in wikipedia. They are the same.

I have found another definition of quaternion slerp in https://users.aalto.fi/~ssarkka/pub/quat.pdf :

`slerp(q1,q2, u) = power(q2#inv(q1), u)#q1        (1)`

where inv(q) stands for the quaternion inverse of q and power(q,u) is its power to u. I have implemented this version of slerp and compared its output with the Q_Slerp ones for randomly generated quaternions. The outputs matched for half the cases. They match whenever:

`` Q_slerp(q1,q2,0.5) ==  Q_slerp(q2,q1,0.5)     (2) ``   

but that is not always true. However, Q_slerp always satisfies to:

`` Q_slerp(q1,q2,0.5) ==  Q_slerp(q1,-q2,0.5)     (3) ``   

and slerp() never does for q1!=q2. The slerp defined by (1) always satisfies (2).

Equation (2) fails whenever the angle between q1 and q2 is greater than 180 degrees. That means Q_Slerp takes the shortest path on the 4D sphere to compute the interpolation and slerp chooses the longest path.

There is an advantage of using slerp instead of Q_slerp: the former is continuous with q1 and q2. Q_slerp shows a discontinuity when the angle between q1 and q2 crosses the 180 degree boundary. slerp() doesn't.

slerp() definition can be simplified exempting the implementation of power() in an expression similar to Q_Slerp :

```
function slerp(q1, q2, u) =
    !is_num(u)? [for (uu=u) slerp(q1,q2,uu)] :
    let(
        q1  = Q_Normalize(q1),
        q2  = Q_Normalize(q2),
        dot = Q_Dot(q1, q2)
    ) 
// the essential difference from Q_Slerp
//    let( 
//      q2  = dot<0? Q_Neg(q2) : q2,
//      dot = dot<0? -dot : dot
//    ) 
//   (dot>0.9995) ? Q_Normalize(q1 + (u * (q2-q1))) :
   (abs(dot)>0.9995) ? Q_Normalize(q1 + (u * (q2-q1))) :
    let(
        dot = constrain(dot,-1,1),
        theta_0 = acos(dot),
        theta   = theta_0 * u,
        s1      = sin(theta) / sin(theta_0),
        s0      = cos(theta) - dot*s1
    )
    (s0 * q1) + (s1 * q2) ;
```

I do not advocate to adopt slerp() instead of Q_Slerp but to offer both. As the codes are very similar, an additional argument of Q_Slerp would define which alternative to follow.

RonaldoCMP, 2020-07-08:
It is not correct to say that slerp() as defined by (1) (where # stands for a quaternion multiplication) always takes the longest path. It takes the path that contains the quaternion:

`normalize((q1+q2)/2)`

Besides, in the codes of both, Q_slerp() and slerp(), variable dot doesn't need to be constrained.

revarbat, 2020-07-09:
Should a combined `Q_slerp()`  function act like `sleep()` or the original `Q_slerp()` by default?

RonaldoCMP, 2020-07-09:
I would keep the current behavior as default. I think that is the most common definition but the continuous alternative may be useful. In fact, the game people have many different quaternion interpolation. The importance of  (both) slerp is due to the constant derivative (speed for them) of the output with the interpolation parameter (time for them). Normalized lerp (called nlerp) has not that property.

I would also define the function:

```function Q_is_quat(q) =  is_vector(q,4) && q!=[0,0,0,0];```

to check input data.

RonaldoCMP, 2020-07-13:
Slerp() is a fast way to interpolate rotations. Unhappily, it does it in quaternion space. So its application requires 3 steps:
1. convert the rotations to interpolate to quaternions;
2. compute the interpolations of the quaternions with slerp at a sequence of parameter values;
3. covert the each interpolated quaternion back to a rotation matrix.

The issue is the step 3 that I found to be expensive compared with the interpolation itself.

An alternative approach would have 2 steps:
a. compute the total rotation matrix to the power of 1/n;
b. compute the n cumulative products of this matrix.

The power can be computed either operating directly with the matrix (using matrix exp and matrix log) or using slerp one time. To keep it simple I have used slerp for that step.

I have compared the two approaches. The run time of the quaternion approach  is more than 10 times the run time of the  alternative approach. That is the repeated operation of slerp and rotation matrix computation of the quaternion approach is much more expensive than the simple repeated operation of a matrix product of the second one. At least in OpenSCAD.

In which applications quaternion would be useful? I have not compared yet the computation of the matrix power by slerp with the computation by exp and log. 

Here are the test codes where it is supposed that q1 and q2 are the given quaternions to interpolate:

// quaternion approach
  qs = Q_Slerp(q1,q2,[for(i=[0:n]) i/n]);
  Mi = Q_Matrix3(Q_inverse(q1));
  Ms = [for(qi=qs) Q_Matrix3(qi)*Mi];

// alternative approach
  qn = qs[0];
  Mn = Q_Matrix3(Q_Slerp(q1,q2,1/n)); 
  Mc = cumprod(Mn, ident(3),n);
 
where cumprod is the cumulative product of a list of matrices and Q_inverse is the inverse of a quaternion:

  function cumprod(M,M0,n) = [for( i=0, P=M0; i<n; i=i+1, P=i<n? M*P: P ) P ];
  function Q_inverse(q) = [-q.x, -q.y, -q.z, q[3] ];

The differences between the matrices Ms[i] and Mc[i] was negligible.

revarbat, 2020-07-14:
The main reason I implemented quaternions was to avoid gimbal lock problems I was seeing when using compounded transformation matrices.

RonaldoCMP, 2020-07-17:
I was naive proposing an alternative way of doing slerp to avoid the discontinuity at 180 degrees. The modified version still has a discontinuity at 360 degrees. It seems there is no way to avoid that.

closed 2020-08-07 by RonaldoCMP