BelfrySCAD/BOSL2 #203: [BUG]

lar3ry, 2020-07-14:
**Describe the bug**
tests/hull.scad gives an error

**Code To Reproduce Bug**
hull.scad

**Expected behavior**
Expected no error

Gives error:
WARNING: len() parameter could not be converted, in file C:/Users/Admin/Documents/OpenSCAD/libraries/BOSL2/geometry.scad, line 554
TRACE: called by 'len', in file C:/Users/Admin/Documents/OpenSCAD/libraries/BOSL2/geometry.scad, line 554.
TRACE: called by 'triangle_area', in file C:/Users/Admin/Documents/OpenSCAD/libraries/BOSL2/hull.scad, line 129.
TRACE: called by '_find_conflicting_segments', in file C:/Users/Admin/Documents/OpenSCAD/libraries/BOSL2/hull.scad, line 105.
TRACE: called by '_hull2d_iterative', in file C:/Users/Admin/Documents/OpenSCAD/libraries/BOSL2/hull.scad, line 96.
TRACE: called by 'hull2d_path', in file C:/Users/Admin/Documents/OpenSCAD/libraries/BOSL2/hull.scad, line 30.
TRACE: called by 'hull', in file _-hull.scad, line 47.
TRACE: called by 'visualize_hull', in file _-hull.scad, line 37.
TRACE: called by 'translate', in file _-hull.scad, line 37.
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
- OpenSCAD Version: 2019.05
tests/hull.scad downloaded just before test

adrianVmariano, 2020-07-25:
I took a look at this, laboriously tracing back through the code to find where the undef comes from, and it originates in `first_noncollinear` which is returning undef when the input is all collinear.   It appears that it previously returned the length of the input (an invalid index), because there is a test for `c==len(points)` that invokes `_hull2d_collinear`.  

So to fix we need to decide what the return will be.  Do we want to change it to [] as per the discussion elsewhere?  Leave it as undef?  I note that the docs don't say anything about what is supposed to happen in this case.  I suppose reverting to previous behavior is also an option, but it doesn't seem right.  Did the dust settle on that discussion with a plan for how to handle these sorts of cases?  

Note also there are two calls to first_noncollinear in hull.scad that need fixing.  These are the only uses of this function in the library.  Does first_noncollinear need to be exposed?

adrianVmariano, 2020-07-25:
Note that first_noncollinear is kind of a weird function.  It takes a list of points, and two indices, and returns the first noncollinear points to the two indices.  Kind of specialized, it seems.   I'm leaning towards moving it into hull.scad and hiding it.

adrianVmariano, 2020-07-31:
Fixed in latest PR.

closed 2020-07-31 by adrianVmariano