BelfrySCAD/BOSL2 #204: include versus use and animation

RonaldoCMP, 2020-07-16:
The strategy of using include<> all over BOSL2 seems to be a way to avoid some bugs of use<>. But it has a severe drawbrack: it spoils animations. Here is a simple example:
```
include<BOSL2/std.scad>
    q1 =  [0.253122, -0.670371, -0.527854, 0.455962] ;
    q2 =  [0.236781, -0.468426, 0.738057, -0.424009];
    n  = 100;
    qs = Q_Slerp(q1,q2,[for(i=[0:n]) i/n]);
    Ms = [for(qi=qs) Q_Matrix4(qi)];
    color("blue") 
      multmatrix(Ms[round(n*$t)]) bar(1);
    %Qrot(q1) bar(); 
    %Qrot(q2) bar(); 
```

If we run this code the animation is very slow, disappointing. But, if instead of `include<BOSL2/std.scad>` we code  `use<BOSL2/std.scad>`, the animation shows dynamism. I can't explain how OpenSCAD deal with those two cases during animations. I just notice the big difference.
I know that a couple of files in BOSL2 have global variable initializations that can degrade a code execution if the files are `used`  instead of `included`. But they are exceptions. For better performance in animations, I think that only these exceptional files should be included.

adrianVmariano, 2020-07-19:
At one point I found that calling a function from a used library was much slower than calling the same function from an included library.  I don't recall any details of how serious this was, but would seem like the main argument against using "use" most of the time.

RonaldoCMP, 2020-07-20:
I have never observed that and a preliminary test I did doesn't confirm it. I know from many reports in the forum and my tests confirmed that a used file with variable initializations will execute the process of initialization each time a function in the file is called. That is the case of the bezier table in beziers.scad, whose computation would be repeated at each call to any file function if it is used. That is an already old OpenSCAD bug.  Except in those cases I prefer `use<>` to `include<>`.

revarbat, 2020-07-20:
I moved everything in BOSL to using include *because* of the re-computing on every call into a used file.  We ran into some pretty ugly performance issues because of it.  We only have a few files with global variable assignments, though.  constants.scad, edges.scad, beliers.scad.  I don't recall if we ever proved that function and module definitions were re-evaluated on every call or not.  If we move to using first-class functions, though, that *definitely* will run afoul of the `use` bug.

revarbat, 2020-07-20:
If we do re-introduce `use`, we still have to include any file that has constants/globals.  We can still require include for std.scad, and have std.scad `use` just the safe files with no constants/globals.

RonaldoCMP, 2020-07-20:
> I don't recall if we ever proved that function and module definitions were re-evaluated on every call or not. 

I could not confirm any re-evaluation using use. I don't know if the file is being loaded with use is re-compiled for each function call like it seems to happen with the include clause.
All this deserves a confirmation test.

> If we move to using first-class functions, though, that _definitely_ will run afoul of the `use` bug.

That is true only if for files initializing variables with functions.

revarbat, 2020-07-21:
The way first-class functions are implemented, you pretty much have to do a lot of things like:

revarbat, 2020-07-21:
```openscad
sqr = function(x) x*x;
```
which is just a function assignment.

adrianVmariano, 2020-07-22:
These are called first-class functions?  Is there any hope that they'll implement a syntax that makes it necessary to (1) redefine all built-in functions as function variables and (2) define many functions using the new function variables?  It seems completely ridiculous that there's no shorthand for function(x) sin(x).  For example, in MATLAB you can write @sin for that.

adrianVmariano, 2020-07-22:
Actually doesn't 
    sqr = function(x) x*x
require that the containing file be included rather than used?  Because if you use it the definition won't make it into the parent workspace.  So their new system is actually going to make it even more necessary to use "include" instead of "use".

RonaldoCMP, 2020-07-22:
I don't see function literal as a substitute to the function definition. Function literal has its place as a parameter for some specific process like a ´compare` function literal in a `sort`. For now, I didn't found much more applications of it. Open to learn.

adrianVmariano, 2020-07-22:
Since I haven't started using function literals it's hard to know exactly what the applications are, but things like applying a function to a list, compare, equation solvers.  There are definitely applications.  And any function that might ever be passed to another function should be defined as a function literal.  So this means we have to redefine all the core functions, I think. 

For best generality we should therefore abandon normal functions entirely and define all functions as function literals.

RonaldoCMP, 2020-07-22:
Although I don't have much experience with function literal, I dare to disagree. First, we don't have much support to function literals yet: there is no way to check whether a function literal has the expect number of arguments and their expected types. That means the input data check is virtually impossible and debug will be hard. Besides, in the rare cases a function literal be valuable, it can be defined from usual functions and the corresponding file should be included. In some cases, the function literal will be defined inside a function or module and the file can be used without harm. That would be the most common case I devise because the average user will not be comfortable using them. 

I think function literals in OpenSCAD now is a non mature experiment. To adopt it along all the library is for BOSL3.

adrianVmariano, 2020-07-22:
Yeah, type checking is a problem indeed.  And definitely function literals are something for BOSL3, not for the current library.  That is the reason I am still running the stable version 2019.05 instead of upgrading.  

I'm not sure who the "average user" is.  I think being able to write something like `map(@sin, data)` is pretty accessible to a user who has gone beyond the basics of unioning or differencing cubes and cylinders.  In fact, it seems like the clumsy notation currently required of `map(function(x) sin(x), data)` makes it less accessible.

revarbat, 2020-07-23:
The issue with function declarations vs function literals comes down to the language oddity that you can't get the function literal definition of a function declaration.  ie: To pass any builtin or declared function, you have to make the equivalent of a lambda.  Off the top of my head:
```openscad
function foo(x) = 6*x*x-4x+7;
bar = function(x) = 6*x*x-4x+7;
function baz(f) = [for (I=[0:1:100) f(x)];
vals1 = baz(foo);  // This is a syntax error.
vals2 = baz(bar);  // This is fine.
vals3 - baz(function(x) foo(x));  // A lambda is fine too.
```
So declaring functions as variables with function literal values is less awkward than requiring lambdas everywhere.

revarbat, 2020-07-23:
Agreed that function literal support is not in the cards for BOSL2.  The OpenSCAD devs haven't really confirmed that function literals are actually going to become supported in mainline.

revarbat, 2020-07-23:
As far as places that function literals would be nice...
- `heightfield()` could use an x,y function literal in obvious ways.
- `grid2d()` could use it as opposed to a region shape for masking.
- `ovoid_spread()` could use it similarly.
- Attachments.scad geometry code could be simplified and optimized a lot.
- `sort()` definitely could use comparator function literals.
There's more, I'm sure.

adrianVmariano, 2020-07-23:
We could write a numerical solver that finds zeros of arbitrary functions, or a function that computes numerical derivatives from a function.

RonaldoCMP, 2020-07-23:
I agree, there are many useful applications for function literals. That is
why it is available in powerful programming languages. But OpenSCAD is not
a regular programming language and its main objective is to produce
parametric geometry. With the current status of function literals we could
produce a vase whose shape is defined by a function. We could produce
the graph of a bivariate function as Revar suggested. I don't think we will
go much further than that in the practical use of the language. It is slow
and cumbersome for numerical applications and it doesn't have any support
for basic things as file read and write and data structure which would be
of great value for implementing computational geometry algorithms.

adrianVmariano, 2020-07-30:
I revisited the performance question and found no difference in the speed of calling a function from a used library vs an included library.  However, I did find a slight difference in calling a used function that called other functions from the library.   In this case, use takes 40% more time than include.   The times are small, though, 2.1 microseconds vs 1.5 microseconds per call.   It's hard to imagine that changes on this time scale can make any difference.

revarbat, 2020-08-19:
I started attempting a rewrite of std.scad to move constants to constants.scad, and have it `use` non-constant library files.  It didn't work.  It will take some careful analysis of the files to have each one `use` all the other library files with functions they reference.

revarbat, 2020-12-01:
I have verified that we can either have slow animations, or incredibly slow function calls.  Until OpenSCAD itself fixes the `use` bug, we really don't have a workable solution here.

closed 2020-12-01 by revarbat