BelfrySCAD/BOSL2 #206: [BUG] typeof

RonaldoCMP, 2020-07-18:
Due to OpenSCAD idiosyncrasies,` typeof()` still has a bug:

```
echo(is_undef([0:NAN:INF])); // false
echo(is_undef([0:"a":INF])); // true
echo(is_undef([0:[]:INF])); // true
echo(is_undef([0:true:INF])); // true
echo();
echo(is_type([0:NAN:INF], "range")); // true
echo(is_type([0:"a":INF], "range"));   // false
echo(typeof([0:NAN:INF]));  // "range"
echo(typeof([0:"a":INF]));  // "undef"
```

The only way I know to catch a valid range is with:

`function is_range(x) = ! is_list(x) && is_finite(x[0]+x[1]+x[2]) ;`

Function `typeof` should be able to return `"none"` or `"invalid"` for invalid ranges.

cf #182

RonaldoCMP, 2020-07-18:
Maybe it is convenient to include also a function like:

`function is_forlist(l) =  is_vector(l) || is_range(l);`

with the `is_range()` definition I proposed.

revarbat, 2020-07-18:
I'm thinking `is_iterable(x)` is a better name for that.

RonaldoCMP, 2020-07-18:
Fine. Any name will be terrible (no pun intended :) ) though it will be useful in places like `select `that fails when `start `is neither a vector nor a valid range.

revarbat, 2020-07-19:
Good catch, BTW.

revarbat, 2020-08-02:
The `typeof()` changes are integrated.  The `is_iterable()` test has not been, yet.

RonaldoCMP, 2020-08-07:
I have introduced _valid_idx() for the purpose of arg validation:

function _valid_idx(idx,imin,imax) =
    is_undef(idx) 
    || ( is_finite(idx) && idx>=imin && idx< imax )
    || ( is_list(idx) && min(idx)>=imin && max(idx)< imax )
    || ( valid_range(idx) && idx[0]>=imin && idx[2]< imax );

It is  defined in arrays.scad as an internal function for now.

revarbat, 2020-08-19:
I'm failing to think of a time I'd want to use `is_iterable()`/`is_forlist()` that I wouldn't really just want `is_list()`, `is_vector()`, and/or `is_range()`.

closed 2020-08-19 by revarbat