BelfrySCAD/BOSL2 #209: sign checking function(s)

adrianVmariano, 2020-07-19:
Add a function and/oror option to is_vector to check for

postive: all entries of list are positive
negative: all entries of list are negative
nonnegative: all entries are >=0
nonpositive: all entries are <=0
zero: all antries are (approximately) zero
nonzero: all entries are nonzero (or epsilon away from zero?)

RonaldoCMP, 2020-07-30:
The `zero` argument has been already implemented in `is_vector` to restrict the kind of vector is allowed. I don't think is a good idea to overload this function (called in hundred places) with more tests. The other tests might be done by a specific function but it is not simple to devise a good interface for it. I thought a bit about and found an intermediate solution: a function which counts the signs of the components of a vector:

```
function sign_count(v, i=0, c=[0,0,0]) =
    i>=len(v)
    ? c
    : sign_count(v, i+1, c=let(s=sign(v[i])) c+(s>0?[1,0,0]:s<0?[0,0,1]:[0,1,0]) );
```

This function result is a vector with 3 numbers `[p,z,n]` with the counts of the positive (p), negative (n) and zero (z) components of v.

An analysis of its output components of we can complete the check:
```
positive    <=> sign_count(v)[0]==len(v)
zero        <=> sign_count(v)[1]==len(v)
negative    <=> sign_count(v)[2]==len(v)
nonpositive <=> sign_count(v)[0]==0
nonnegative <=> sign_count(v)[2]==0
```

revarbat, 2020-07-30:
To me `sign_count()` makes for non-obvious checks.

I really think this should be handled by two new functions:
- `positive(x)` returns true if scalar `x` or all components of vector `x` are > 0.
- `negative(x)` returns true if scalar `x` or all components of vector `x` are < 0.

Mayyyybe also a `is_zero(x)` function.  The expression `norm(x)>0` is pretty short and clear, but it doesn't approximate zero.

You can just do `!negative(x)` for non-negative, etc.

adrianVmariano, 2020-07-31:
I agree that sign_count is kind of obscure.  I don't think I would ever actually want to check that the sign count was some value other than zero or len(v), and it's verbose.   Perhaps sign_count is the internal function used to define other user facing functions?   Though that's probably less efficient than a direct implementation.  

I disagree that adding other tests to is_vector results in a complex interface.  I think `is_vector(v, positive=true)` is pretty short and clean.   And then `is_vector(v,negative=false)` would handle the case of nonnegative.  And similarly for `positive=false` and `negative=true`.   

But if the general theory is that we need separate functions then we have to think about the interface.  

If we suppose that `positive(v)` means all entries of v > 0 then `!positive(v)` is NOT the non-positive inverse, because it means at least one entry is non-negative, not all of them.  Similarly if `negative(v)` means all entries are negative than `!negative(v)` means at least one entry is non-negative....not all of them.   So you're going to have to either add `nonnegative()` and `nonpositive()` to this setup, or think of a different interface.   There's also a need for `is_nonzero` presumably, which would check that all entries are not approximately zero.  

This was why I proposed `check_sign` if we want a separate function, because then you can make it work where `check_sign(v,positive=true)` requires all values positive and `check_sign(negative=false)` requires all values non-negative.

revarbat, 2020-07-31:
You have a point about nonnegative/nonpositive.

revarbat, 2020-07-31:
Honestly, having `positive(x)`, `negative(x)`, `nonnegative(x)`, `nonpositive(x)`, and `is_zero(x)` makes for more readable code than `check_sign(x,nonnegative=true)`.

adrianVmariano, 2020-07-31:
You don't need `check_sign(x,nonnegative=true)` because you can use `check_sign(x,negative=false)` for that.   But It is still less readable than `nonnegative(x)`, I suppose.

closed 2020-08-27 by revarbat