BelfrySCAD/BOSL2 #212: List manipulation and strings

RonaldoCMP, 2020-07-22:
Most list manipulation in arrays.scad accept strings as input. However the returned value is odd when the input is a string. For example `reverse` does not reverse the string; instead it returns the reversed list of the input string characters. An useful function missing in arrays.scad is:

```
function str_concat(list) =
  assert( is_list(list), "Invalid list.")
  [for(i=len(list)-1, s=list[i]; 
	   i>=0; 
		   i=i-1, s=str(s,list[i])) if(i==0) s ][0]; 
```
This function concatenates the input list using `str` instead of concat and returns a string, not a list.

```
   str_concat(["a","b","cd"]); // Returns "abcd"
   str_concat(["a",0,"cd"]);   // Returns "a0cd"
   str_concat([1,2,3]);        // Returns "123"
```
With the help of `str_concat`, function `reverse` could reverse strings the way it is expected returning a string:

```
function reverse(list) =
    is_list(list) ? [ for (i = [len(list)-1 : -1 : 0]) list[i] ] :
		is_string(list) ? str_concat([ for (i = [len(list)-1 : -1 : 0]) list[i] ])
    : assert(is_list(list) || is_string(list), "Improper list or string.");

```
Many other functions in arrays.scad could similarly return strings (and not a list of characters) to string inputs.

RonaldoCMP, 2020-07-22:
I just found that `str_join` in strings.scad is similar to `str_concat` above. Although strings.scad has no reverse function for strings it could be performed with `reverse` followed by `str_join`.

adrianVmariano, 2020-07-24:
I'm inclined to think that the list manipulation functions should not operate on strings.  I see that Revar specifically allows strings as input, but as you note, the result is not what you'd expect.  It seems better to limit the array.scad functions to operating on true lists, with an assert for string inputs, and to put string operations in strings.scad.   Reversing a string is not a very useful function---I never needed it---hence the lack in the string library.

revarbat, 2020-08-26:
I had originally allowed for list functions to operate on strings because OpenSCAD itself allowed for it.  Since you can rejoin lists with `str_join()`, though, I don't see much reason to not allow it. While list functions behave oddly with strings, they behave consistently with how OpenSCAD itself behaves with strings.

adrianVmariano, 2020-08-27:
The main reason not to allow it is that the output isn't the right thing.  I think the argument that it mimics stupid OpenSCAD behavior is sort of weak.   It seems to me that if strings are to be supported then those functions should all call str_join when the input is a string so that the output is a string instead of a list of characters.

revarbat, 2020-09-29:
Fixed most list functions to return strings from string input.
Implemented in e3ccf482fadcc2e7f5a8d5021252d7136855b367

closed 2020-09-29 by revarbat