BelfrySCAD/BOSL2 #214: vmul()

RonaldoCMP, 2020-07-24:
`vmul` has a bug:

```
vmul([1,2,3],[1,2]) == [1,4,undef]
vmul([1,2],[1,2,3]) == [1,4]
```

We have three alternatives to avoid the bug: 
1. consider both calls an error (my preference) 
2. `vmul `clip the longer vector before the multiplication as OpenSCAD does with the sum of vectors (and that is trouble!); so, for both calls above it would return `[1,4]`; the operation would be commutative;
3. consider an error when the length of the first vector is greater than the length of the second and proceed as in 2; the operation would be non commutative as it is now but no `undef `would be generated.

Alternative 1 has side effects in scale() as my last PR denounces: scales() and their derivatives doesn't compare the dimensions of the scale factors with the dimensions of the points to scale.  The regression tests of shapes.scad and shapes2d.scad also fails with alternative 1. 

It seems that alternatives 2 and 3 would pass all regression tests because no undef seems to be generated now. However, they seem odd and they are more _ad hoc_ solution to avoid the review of other codes.

adrianVmariano, 2020-07-24:
I think that alternative 1 is correct, mismatched vectors should be an error, and where the code fails the code needs to be fixed.

RonaldoCMP, 2020-07-27:
I have found two places were `vmul() `of dissimilar length vectors is done.
1. in transform.scad, the function `scale()` and the following check wrongly (or not at all) the vector dimensions ;
2. in shapes2d.scad, the function `rect()` calls `vmul()` mixing a 3D vector `anchor `with the 2D vector `size`.

I suggest that other similar functions be reviewed to check if they have the same irregular use of `vmul`.

adrianVmariano, 2020-07-27:
There are a total of 40 calls to vmul in the library that occur in attachments, beziers, distributors, math, partitions, polyhedra, primitives, shapes2d, shapes, and transforms.  Which files still need to be checked?

adrianVmariano, 2020-07-28:
I looked over the calls and everything looks good in beziers, distributors, partitions, polyhedra, and primitives---though these files need error handling.   I didn't understand attachments well enough and wasn't sure if you'd already looked at the two shapes files.   I think we should push the fix to vmul out.

adrianVmariano, 2020-07-31:
What needs to happen so we can close this one?

revarbat, 2020-07-31:
I need to check attachments code at least.

revarbat, 2020-08-03:
Fixed in a05509dc6d2ea2a3fefc5cd1e9ab172713b232d6

closed 2020-08-03 by revarbat