BelfrySCAD/BOSL2 #217: consistent and numerical lists

RonaldoCMP, 2020-07-26:
The function `is_consistent()` doesn't ensure the input list contains just numbers. For instance, `is_consistent([ [1,undef], [2,"a"] ])` is true. It ensures that if we substitute each number in the list by 1 and any other simple value by 0, all list items will be equal. The same happens with `same_shape()`. 

Some functions, like `deriv()`, doesn't require just list of numbers but that the list be consistent and containing just numbers. To check the later requires recursion to confirm that each individual item is a number. That is my proposal to such a test:
```

function is_numerical(list) =
    is_finite(list) 
    || is_vector(list)
    || is_matrix(list)
    || ( is_list(list) 
         && []==[for(vi=list) if(! is_numerical(vi)) 0]
       );
```
Note that `is_numerical()` doesn't ensure consistence and it is usually more expensive than the consistence test.

I will insert this definition in math.scad and the input data checks in `deriv() `using it.

RonaldoCMP, 2020-07-27:
I have found a bit faster and simpler code for `is_numerical`. It doesn't call other defined test functions in order to reduce the overload in input data checks. 

```
function is_numerical(list,void=true) =
    ! is_list(list) ? is_num(0*list)
    : ( ( let(v = list*list[0]) is_num(0*(v*v)) ) // true just for vectors and matrices
        || []==[for(vi=list) if(! is_numerical(vi)) 0] );

```
Another useful test function is the one that approves just consistent numerical arrays. I call it is_addable, for lack of a better name, because any two of its first level entries of the array may be numerically added without a flaw:

```
function is_addable(list) = // consistent and numerical
    is_list_of(list,list[0])
    &&  ( ( let(v = list*list[0]) is_num(0*(v*v)) ) 
        || []==[for(vi=list) if(! is_numerical(vi)) 0] );

```
These function pass the following tests:

```
assert(is_numerical([]));
assert(is_addable  ([]));
assert(is_numerical([[1]]));
assert(is_addable  ([[1]]));
assert(is_numerical([[1,[]]]));
assert(is_addable  ([[1,[]]]));
assert(is_numerical([[[1],2],[[0],2]]));
assert(is_addable  ([[[1],2],[[0],2]]));
assert(is_numerical([[1,[2]],[2,[3]],[4,[4]]]));
assert(is_addable  ([[1,[2]],[2,[3]],[4,[4]]]));
assert(is_numerical([[[1],2],[[0],[]]]));
assert(! is_addable([[[1],2],[[0],[]]]));
assert(!is_numerical([[[1],2],[[0],[""]]]));
assert(! is_addable ([[[1],2],[[0],[""]]]));
```

RonaldoCMP, 2020-07-30:
Just to close this issue, Adrian's solution was simpler than the one I had proposed: change is_consistent and same_shape to approve just numerical lists:

```
function is_consistent(list) =
  is_list(list) && is_list_of(list, _list_pattern(list[0]));

function _list_pattern(list) =
  is_list(list) 
  ? [for(entry=list) is_list(entry) ? _list_pattern(entry) : 0]
  : 0;

function same_shape(a,b) = _list_pattern(a) == b*0;
```

RonaldoCMP, 2020-07-31:
It should be noted that according to this new definition an empty list is consistent because

```
_list_pattern(undef)==0 
is_list_of([],0)==true
```

I agree with this result. Does anyone disagree?

When the list is empty, `list[0]` is (currently) `undef` but `_list_pattern` can deal with this. If this is considered unsafe, the definition should change:

```
function is_consistent(list) =
  is_list(list) 
  && ( len(list)==0 || is_list_of(list, _list_pattern(list[0])) );
```

If it is safe, `is_consistent` may be simplified:

`function is_consistent(list) = is_list_of(list, _list_pattern(list[0]));`

adrianVmariano, 2020-07-31:
It's a little hard to guess what's safe at the moment, though it seems that there are some strong forces in favor of keeping list[0] always safe.  

It seems to me that empty lists evaluating as consistent is reasonable.   Maybe we should assume it's safe, and make sure it's tested by the regression tests.  It'll be then easy to catch and fix it it ever becomes unsafe.

adrianVmariano, 2020-07-31:
Actually even if list[0] is safe, undef*0 may not be.

RonaldoCMP, 2020-08-01:
Well, if undef*0 gets unsafe, a lot of our basic check functions will break:

is_finite
is_list_of
is_consistent
is_vector
is_matrix

All those functions multiply an input data by 0 to detect non numerical values, undef included.

adrianVmariano, 2020-08-01:
I actually think it will be OK.  The undef may be caught earlier.   Or if not, multiplying by 0 will reveal the non-numerical values by generating an error at that point.   It's a bit unclear what will happen:  if 0*undef = undef but no error yet, and then you compare that result to 0, so undef==0...well, that may also be legal because it seemed the idea was to allow undefs to appear in conditions but only generate an error when it is "used".  It's not clear what that means exactly.

RonaldoCMP, 2020-08-01:
It is not so simple. An assert that the input list is a vector or a string would have to care the order of the checks:

assert( is_vector(v) || is_string(v) )

will break if v is a string. Similarly an extra care would be needed in non assert conditionals involving those functions.

adrianVmariano, 2020-08-02:
No, is_vector(string) is false because is_list(string) is false---the first test in is_vector.  No problem there.  

I'm really not sure that there is any problem.  It sounds like you can run 0*undef and nothing fails yet.  You can run -0==undef and that's valid.  So where will it fail?

adrianVmariano, 2020-08-02:
If 0*undef does fail then there is a problem in the case where you want to allow a vector or matrix or alternatively a list containing strings or booleans.  I think it's a kind of weird case.  So in that case you'd need to test for the list of strings first, yes.

This seems like it's in the category of we'll figure it out once it arrives and we know what has actually arived.  

Looks like someone commented out the is_list check in is_consistent.  I think this is resolved, at least for the present.

closed 2020-08-02 by adrianVmariano