BelfrySCAD/BOSL2 #222: [BUG] in_list

RonaldoCMP, 2020-07-30:
According with the `in_list` docs:

```
// Description: Returns true if value `val` is in list `list`. When `val==NAN` the answer will be false for any list.
// Arguments:
//   val = The simple value to search for.
//   list = The list to search.
//   idx = If given, searches the given subindexes for matches for `val`.
// Example:
//   in_list("bar", ["foo", "bar", "baz"]);  // Returns true.
//   in_list("bee", ["foo", "bar", "baz"]);  // Returns false.
//   in_list("bar", [[2,"foo"], [4,"bar"], [3,"baz"]], idx=1);  // Returns true.
```

it should return `true` for:

`in_list(val=2, list=[[0,1],[1,2]], idx=[0,1])`

however its result is `false`.

The point is that function `search` doesn't work for a list of indices just for a single index.

We have two alternatives:
1. restrict `idx` to be either undef or a number
2. when `idx` is a list of numbers call `search` for each element in `idx`.

I could not find any place in the library where `in_list` is called with a list `idx`.

revarbat, 2020-07-30:
I don't think I ever considered a non-scalar `idx` value for this.

Would it be useful to allow for a vector `idx`?  What would it mean?  I can see either of:
- "search for val in any of these subindexes" `any([for (i=idx) ...])` or
- "search for val in these recursive subindexes" `subindex(subindex(list,idx[0]),idx[1]...)`

RonaldoCMP, 2020-07-30:
If you have never considered non-scalar `idx`, then we just need to correct the doc and include an argument check on `idx`.

closed 2020-08-07 by RonaldoCMP