BelfrySCAD/BOSL2 #231: Missing regression tests

RonaldoCMP, 2020-08-10:
Still missing regression test files:

- test_attachements.scad
- test_beziers.scad
- test_bosl1_compat.scad
- test_bottlecasp.scad
- test_constants.scad (?)
- test_distributors.scad
- test_hingesnaps.scad
- test_involute_gears.scad
- test_joiners.scad
- test_knurling.scad
- test_masks.scad
- test_metric_screws.scad
- test_nema_steppers.scad
- test_partitions.scad
- test_paths.scad
- test_phillips_drive.scad
- test_regions.scad
- test_rounding.scad
- test_screws.scad
- test_sliders.scad
- test_threading.scad
- test_torx_drive.scad
- test_triangulation.scad
- test_walls.scad
- test_wiring.scad

The regression tests:

- hull.scad
- polyhedra.scad

should be called

- test_hull.scad
- test_polyhedra.scad

revarbat, 2020-08-10:
This is redundant with issue #102. The issue is a long task, in progress.

RonaldoCMP, 2020-08-11:
My intention is to have a picture of what has to be done. As soon as a test file is created, I would edit the above list. Another suggestion is to begin each test file with the test calls in the order the functions are defined in the main file so that it would be easier to check if all function has been checked. I know it is a long task.

revarbat, 2020-08-12:
Makes sense. You can edit your list comment, and use the markup `[ ]` and `[x]` to get clickable/checkable checkboxes, so we can check them off as they are done.  The percentage checked will also be shown in the project level.
```markdown
- [ ] Unchecked
- [x] Checked
```
- [ ] Unchecked
- [x] Checked

adrianVmariano, 2020-08-12:
Note that hull.scad and polyhedra.scad aren't regression tests, but more like demos.  I wrote test_hull.scad.   Since we don't have a way of doing regression tests for modules, also I think many of those files aren't testable.

RonaldoCMP, 2020-08-12:
> Makes sense. You can edit your list comment, and use the markup `[ ]` and `[x]` to get clickable/checkable checkboxes, so we can check them off as they are done. The percentage checked will also be shown in the project level.

There is another issue: how complete is each test. I am working on geometry.scad and found that around 40% of the functions have no test in the test_geometry,scad.

revarbat, 2020-08-12:
We have a script, `scripts/func_coverage.py` that lists all public functions that have no regression tests, by file. It would be possible to make PR submissions annotate themselves with that.

revarbat, 2020-08-12:
Current output of func_coverage.py:
```
WARNING!!! Function count_true re-defined at math.scad:1011
           Previously defined at math.scad:1001
WARNING!!! Function poly_mult re-defined at math.scad:1237
           Previously defined at math.scad:1220
WARNING!!! Function sortidx re-defined at arrays.scad:860
           Previously defined at arrays.scad:838
NOT COVERED:
  quaternions.scad: 1 uncovered functions
    Q_is_quat
  common.scad: 1 uncovered functions
    is_finite
  wiring.scad: 2 uncovered functions
    hex_offset_ring
    hex_offsets
  hull.scad: 2 uncovered functions
    hull2d_path
    hull3d_faces
  screws.scad: 2 uncovered functions
    screw_info
    thread_specification
  edges.scad: 3 uncovered functions
    corners
    is_corner_array
    normalize_corners
  polyhedra.scad: 3 uncovered functions
    regular_polyhedron_info
    stellate_faces
    trapezohedron
  skin.scad: 5 uncovered functions
    associate_vertices
    path_sweep
    slice_profiles
    subdivide_and_slice
    sweep
  torx_drive.scad: 5 uncovered functions
    torx_depth
    torx_inner_diam
    torx_outer_diam
    torx_rounding_radius
    torx_tip_radius
  nema_steppers.scad: 6 uncovered functions
    nema_motor_plinth_diam
    nema_motor_plinth_height
    nema_motor_screw_depth
    nema_motor_screw_size
    nema_motor_screw_spacing
    nema_motor_width
  vnf.scad: 6 uncovered functions
    vnf_bend
    vnf_centroid
    vnf_compact
    vnf_quantize
    vnf_validate
    vnf_volume
  triangulation.scad: 7 uncovered functions
    face_normal
    find_convex_vertex
    is_only_noncolinear_vertex
    normalize_vertex_perimeter
    point_in_ear
    triangulate_face
    triangulate_faces
  attachments.scad: 8 uncovered functions
    anchorpt
    attach_geom
    attach_geom_2d
    attach_geom_size
    attach_transform
    attachment_is_shown
    find_anchor
    reorient
  involute_gears.scad: 12 uncovered functions
    adendum
    base_radius
    bevel_pitch_angle
    circular_pitch
    dedendum
    diametral_pitch
    gear2d
    gear_tooth_profile
    module_value
    outer_radius
    pitch_radius
    root_radius
  metric_screws.scad: 12 uncovered functions
    get_metric_bolt_head_height
    get_metric_bolt_head_size
    get_metric_iso_coarse_thread_pitch
    get_metric_iso_fine_thread_pitch
    get_metric_iso_superfine_thread_pitch
    get_metric_jis_thread_pitch
    get_metric_nut_size
    get_metric_nut_thickness
    get_metric_socket_cap_diam
    get_metric_socket_cap_height
    get_metric_socket_cap_socket_depth
    get_metric_socket_cap_socket_size
  regions.scad: 14 uncovered functions
    check_and_fix_path
    cleanup_region
    close_region
    difference
    exclusive_or
    intersection
    linear_sweep
    offset
    point_in_region
    region_faces
    region_path_crossings
    split_nested_region
    split_path_at_region_crossings
    union
  rounding.scad: 15 uncovered functions
    angle_between_lines
    bezier_patch_degenerate
    offset_stroke
    os_chamfer
    os_circle
    os_flat
    os_pointed
    os_profile
    os_round
    os_smooth
    os_teardrop
    round_corners
    rounded_prism
    smooth_path
    vnf_tri_array
  paths.scad: 23 uncovered functions
    assemble_a_path_from_fragments
    assemble_path_fragments
    decompose_path
    path3d_spiral
    path_closest_point
    path_curvature
    path_cut
    path_length
    path_length_fractions
    path_normals
    path_pos_from_end
    path_pos_from_start
    path_segment_lengths
    path_self_intersections
    path_subselect
    path_tangents
    path_torsion
    path_trim_end
    path_trim_start
    points_along_path3d
    resample_path
    split_path_at_self_crossings
    subdivide_path
  geometry.scad: 25 uncovered functions
    align_polygon
    circle_circle_tangents
    closest_point_on_plane
    line_ray_intersection
    plane_from_normal
    plane_from_polygon
    plane_intersection
    plane_line_angle
    plane_line_intersection
    plane_offset
    plane_point_nearest_origin
    plane_transform
    point_left_of_segment2d
    points_are_collinear
    points_are_coplanar
    polygon_is_convex
    polygon_line_intersection
    polygon_normal
    ray_closest_point
    ray_intersection
    ray_segment_intersection
    reindex_polygon
    split_polygons_at_each_x
    split_polygons_at_each_y
    split_polygons_at_each_z
  beziers.scad: 25 uncovered functions
    bezier_close_to_axis
    bezier_curvature
    bezier_curve
    bezier_derivative
    bezier_offset
    bezier_patch
    bezier_patch_flat
    bezier_patch_points
    bezier_path_closest_point
    bezier_path_length
    bezier_path_point
    bezier_points
    bezier_polyline
    bezier_segment_closest_point
    bezier_segment_length
    bezier_surface
    bezier_tangent
    bezier_triangle_point
    fillet3pts
    fillet_path
    is_patch
    is_rectpatch
    is_tripatch
    patch_reverse
    path_to_bezier
Total coverage: 401 of 578 functions (69.38%)
```

RonaldoCMP, 2020-08-16:
Another issue regarding the regression tests is to assure all arg options are considered in the tests of a function. For instance, test_find_circle_2tangents() in test_geometry.scad miss any test with with arg tangent=true.

adrianVmariano, 2020-08-16:
I think this is part of the broader problem of whether tests cover all cases.  I am dubious that we can automate assessment of whether regression tests are complete.  It's something we just have to check by hand.  When I was adding requested regression tests to test_math I noticed several other regression tests that missed edge cases...and I actually found some bugs when I added the missing edge cases.

revarbat, 2020-08-19:
Updated run of scripts/func_coverage.py:
```
NOT COVERED:
  quaternions.scad: 1 uncovered functions
    Q_is_quat
  common.scad: 1 uncovered functions
    is_finite
  wiring.scad: 2 uncovered functions
    hex_offset_ring
    hex_offsets
  hull.scad: 2 uncovered functions
    hull2d_path
    hull3d_faces
  screws.scad: 2 uncovered functions
    screw_info
    thread_specification
  edges.scad: 3 uncovered functions
    corners
    is_corner_array
    normalize_corners
  polyhedra.scad: 3 uncovered functions
    regular_polyhedron_info
    stellate_faces
    trapezohedron
  skin.scad: 5 uncovered functions
    associate_vertices
    path_sweep
    slice_profiles
    subdivide_and_slice
    sweep
  torx_drive.scad: 5 uncovered functions
    torx_depth
    torx_inner_diam
    torx_outer_diam
    torx_rounding_radius
    torx_tip_radius
  nema_steppers.scad: 6 uncovered functions
    nema_motor_plinth_diam
    nema_motor_plinth_height
    nema_motor_screw_depth
    nema_motor_screw_size
    nema_motor_screw_spacing
    nema_motor_width
  vnf.scad: 6 uncovered functions
    vnf_bend
    vnf_centroid
    vnf_compact
    vnf_quantize
    vnf_validate
    vnf_volume
  triangulation.scad: 7 uncovered functions
    face_normal
    find_convex_vertex
    is_only_noncolinear_vertex
    normalize_vertex_perimeter
    point_in_ear
    triangulate_face
    triangulate_faces
  attachments.scad: 8 uncovered functions
    anchorpt
    attach_geom
    attach_geom_2d
    attach_geom_size
    attach_transform
    attachment_is_shown
    find_anchor
    reorient
  involute_gears.scad: 12 uncovered functions
    adendum
    base_radius
    bevel_pitch_angle
    circular_pitch
    dedendum
    diametral_pitch
    gear2d
    gear_tooth_profile
    module_value
    outer_radius
    pitch_radius
    root_radius
  metric_screws.scad: 12 uncovered functions
    get_metric_bolt_head_height
    get_metric_bolt_head_size
    get_metric_iso_coarse_thread_pitch
    get_metric_iso_fine_thread_pitch
    get_metric_iso_superfine_thread_pitch
    get_metric_jis_thread_pitch
    get_metric_nut_size
    get_metric_nut_thickness
    get_metric_socket_cap_diam
    get_metric_socket_cap_height
    get_metric_socket_cap_socket_depth
    get_metric_socket_cap_socket_size
  regions.scad: 14 uncovered functions
    check_and_fix_path
    cleanup_region
    close_region
    difference
    exclusive_or
    intersection
    linear_sweep
    offset
    point_in_region
    region_faces
    region_path_crossings
    split_nested_region
    split_path_at_region_crossings
    union
  rounding.scad: 16 uncovered functions
    angle_between_lines
    bezier_patch_degenerate
    offset_stroke
    offset_sweep
    os_chamfer
    os_circle
    os_flat
    os_pointed
    os_profile
    os_round
    os_smooth
    os_teardrop
    round_corners
    rounded_prism
    smooth_path
    vnf_tri_array
  paths.scad: 23 uncovered functions
    assemble_a_path_from_fragments
    assemble_path_fragments
    decompose_path
    path3d_spiral
    path_closest_point
    path_curvature
    path_cut
    path_length
    path_length_fractions
    path_normals
    path_pos_from_end
    path_pos_from_start
    path_segment_lengths
    path_self_intersections
    path_subselect
    path_tangents
    path_torsion
    path_trim_end
    path_trim_start
    points_along_path3d
    resample_path
    split_path_at_self_crossings
    subdivide_path
  geometry.scad: 25 uncovered functions
    align_polygon
    circle_circle_tangents
    closest_point_on_plane
    line_ray_intersection
    plane_from_normal
    plane_from_polygon
    plane_intersection
    plane_line_angle
    plane_line_intersection
    plane_offset
    plane_point_nearest_origin
    plane_transform
    point_left_of_segment2d
    points_are_collinear
    points_are_coplanar
    polygon_is_convex
    polygon_line_intersection
    polygon_normal
    ray_closest_point
    ray_intersection
    ray_segment_intersection
    reindex_polygon
    split_polygons_at_each_x
    split_polygons_at_each_y
    split_polygons_at_each_z
  beziers.scad: 25 uncovered functions
    bezier_close_to_axis
    bezier_curvature
    bezier_curve
    bezier_derivative
    bezier_offset
    bezier_patch
    bezier_patch_flat
    bezier_patch_points
    bezier_path_closest_point
    bezier_path_length
    bezier_path_point
    bezier_points
    bezier_polyline
    bezier_segment_closest_point
    bezier_segment_length
    bezier_surface
    bezier_tangent
    bezier_triangle_point
    fillet3pts
    fillet_path
    is_patch
    is_rectpatch
    is_tripatch
    patch_reverse
    path_to_bezier
Total coverage: 400 of 578 functions (69.20%)
```

RonaldoCMP, 2020-08-19:
> Makes sense. You can edit your list comment, and use the markup `[ ]` and `[x]` to get clickable/checkable checkboxes, so we can check them off as they are done. The percentage checked will also be shown in the project level.
> 
> ```gfm
> - [ ] Unchecked
> - [x] Checked
> ```
> 
> * [ ]  Unchecked
> * [x]  Checked

Done

adrianVmariano, 2020-08-20:
Note that hull should actually be fully covered because the main hull calls hull2d and hull3d depending on the dimension of the input.  Should we write dummy tests?  Recopy the tests?

closed 2026-09-23 by adrianVmariano