BelfrySCAD/BOSL2 #232: [BUG] Warnings generated by involute_gears.scad

boxcarmib, 2020-08-10:
**Describe the bug**
the "example train" code in the source file itself first returns a cascade of the following warnings...

WARNING: matrix*matrix requires left operand column count to match right operand row count (2 != 14) in file ../../../../OpenSCAD/_libraries/current/BOSL2/vectors.scad, line 38 
WARNING: undefined operation (number * undefined) in file ../../../../OpenSCAD/_libraries/current/BOSL2/vectors.scad, line 38 

**Code To Reproduce Bug**
The code is in the source file itself

**Expected behavior**
I expected it to run without warnings. It appears to preview and render successfully but the warnings make other debugging efforts difficult to trace.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version:
version 2020.08.08 (git 93c1c211) 

- Other libraries used:

adrianVmariano, 2020-08-14:
This behavior is due to new handling of undefined operations in the bleeding edge dev snapshot 2020.08.08.  If you go back to an older version it will run without errors.  For now I recommend running with an older version, as this change will probably produce tons of issues throughout BOSL2 and we don't know yet what the final form is for this  new feature.  

As an aside, the example produces a rack gear that isn't properly aligned with the round gear it is supposed to mesh with, so something is wrong there.  

The handling of undef has changed, and not quite the way it was advertised on the forum.  I thought that undef operations were going to be warning-free until you tried to operate on the undef.  But in fact the more "standard" behavior occurs.  As soon as you set C = A*B where the operands are incompatible, you get a warning.   This means that is_vector and various other fundamental functions are broken.

revarbat, 2020-10-05:
Verified fixed.

closed 2020-10-05 by revarbat