BelfrySCAD/BOSL2 #258: Module for generating shell 3D models

olemis, 2020-09-02:
**Is your feature request related to a problem? Please describe.**
A common pattern in 3D models consists of a base 3D shape with vertical shells (walls) like if base was extruded but hollow in the inside . It seems there is no way to do so out of the box.

**Describe the solution you'd like**
An operator that receives as input the base 3D model and outputs the same object augmented with shell walls.

**Describe alternatives you've considered**
Doing this by hand might be a quite annoying repetitive task.

**Example Code**
Usage should be like this

```
shell(<width>, <height>) <3D mnodel>
```

For instance

```
shell(0.3, 10)
    hull() {
           translate(3,3,0)
               sphere(r=3);
           translate(7,3,0)
               sphere(r=3);
           translate(3,12,0)
               sphere(r=3);
           translate(7,12,0)
               sphere(r=3);
    }
```

... should render ...

[image]

[image]

A similar example for chamfer

```
module pyramid(height) {
    polyhedron(points=[[0, 0, height], [height, 0, 0], [0, -height, 0], [-height, 0, 0], [0, height, 0]], faces=[[0, 1, 2], [0, 2, 3], [0, 3, 4], [0, 4, 1], [1,2,3,4]]);
}

shell(2, 20)
    hull() {
        translate([-20, -20])
            mirror([0,0,1]) pyramid(4);
        translate([20, -20])
            mirror([0,0,1]) pyramid(4);
        translate([-20, 20])
            mirror([0,0,1]) pyramid(4);
        translate([20, 20])
            mirror([0,0,1]) pyramid(4);
    }
```

[image]

[image]

**Additional context**
The example explained above arose while discussing a real-life application for OTTO-DIY [body](https://wikifactory.com/+OttoDIY/otto-diy/file/3Dprint/OttoDIY_body_V10.stl) and [head](https://wikifactory.com/+OttoDIY/otto-diy/file/3Dprint/OttoDIY_head_V10.stl).

olemis, 2020-09-02:
I'll contribute the solution soon , What would be the best file to include it in ? `walls.scad` ?

revarbat, 2020-09-07:
mutators.scad, actually.

olemis, 2020-09-14:
> mutators.scad, actually.

done in 8a83685

olemis, 2020-10-01:
`shell` name clashes with [scad-utils](https://github.com/openscad/scad-utils) module of the same name.

revarbat, 2020-11-30:
This has an implementation that is both fairly simple, yet unusual in specific needs.  If you have a 3D shape for the bottom of the piece, aligned below the XY plane, you could use a module like:
```openscad
module cup(h=10, wall=3, convexity) {
    linear_extrude(height=h, convexity=convexity)
        shell2d(wall) projection() children();
    children();
}
```

closed 2020-12-17 by adrianVmariano