BelfrySCAD/BOSL2 #262: sqr() and sqr_entries()?

adrianVmariano, 2020-09-03:
Would it make sense to change sqr() to sqr_entries() and introduce sqr() to be just its argument times itself, so it can work on matrices?

revarbat, 2020-09-24:
Seems to me that `sqr(v)` should vary by type of `v`:
- Where v is scalar: `v*v`
- Where v is a vector: `vmul(v,v)`
- Where v is a matrix: `v*v`

All of those are so simple, I'm now questioning the need for `sqr()` at all.

adrianVmariano, 2020-09-25:
`sqr()` is absolutely needed.   Yes, it's simple, but I don't want to write

    (a+b-2*x+3-y)*(a+b-2*x+3-y)

revarbat, 2020-09-25:
Implemented in 1435100e37d18bdd9a932055cd91a64435686d01

adrianVmariano, 2020-09-26:
I was just noticing that if v is a vector then v*v is the sum of squares.   This makes it seem like sum_of_squares should go away, and sqr(v) should compute v*v, and my original idea of sqr_entries introduced to square the entries of a list.   Thoughts?  Note sum_of_squares should be rewritten as v*v presumably if it is retained.   Any idea how often sqr() is used on a vector in the existing code?

revarbat, 2020-09-28:
I would have said I don't recall actually *using* `sqr()` anywhere else in the library, but there actually seems to be a fair number of places it's used.  I don't *think* it's used on anything but scalars anywhere in the library, though.  The only places I'm unsure of are in:
- `bezier_curvature()`
- `path_curvature()`

The semantics of what squaring a vector means isn't clear to me.  The vector `v*v` is specifically a dot product, which is also the sum of squares.  But an element-wise squaring seems as valid to me.   I'm willing to just make it `function sqr(x) = x*x;`, though, for elegance.  But wasn't that what we started with?

BTW, I noticed a couple places where we use the pattern `sqrt(sqr(x)+sqr(y))`, which is the same as `norm([x,y])`.

revarbat, 2020-09-28:
Given that the library uses element-wise squaring absolutely nowhere, and it can be done by `[for (x=vec) x*x]`, I don't really see a need for `sqr_entries()`.

closed 2020-09-29 by revarbat