BelfrySCAD/BOSL2 #269: Emboss base object using 3D pattern

olemis, 2020-09-18:
**Is your feature request related to a problem? Please describe.**
There is no mutator to implement a quite common pattern consisting in adding and subtracting (at the same time) a pattern out of multiple base objects ... similar to stamping or [paper embossing](https://en.wikipedia.org/wiki/Paper_embossing#Embossing_types).

**Describe the solution you'd like**
The workflow should be as follows:

- Design a 3D mask model for the repetitive embossing pattern based on the following rules
  * Mask parts in Z+ sub-space will be added to base model
  * Mask parts in Z- sub-space will be subtracted out of base object
- Design multiple base objects 
- Apply mask upon them.

**Describe alternatives you've considered**
This can be done by implementing modules one-by-one for each particular case , but this approach is tedious and cumbersome. A data driven solution can be implemented once for multiple use cases.

**Example Code**

Mask 👇
[image]

```
emboss() {
    mask();
    cube([6, 10, 3], center=true);
}
```

Result 👇
[image]

This is a real-life example.

**Additional context**
A few more variations would be nice to have , e.g. plane in direction other than Z+ and base object located at a point in space.

olemis, 2020-09-18:
I'll submit an initial patch soon.

revarbat, 2020-11-30:
```openscad
module emboss() {
    assert($children==2);
    difference() {
        children(0);
        bottom_half() children(1);
    }
    top_half() children(1);
}
```

revarbat, 2020-11-30:
This isn't exactly embossing. Not sure what to call it, though.

revarbat, 2020-12-01:
The `top_half()` and `bottom_half()` modules sometimes need a size value.
Also, `difference()` and `union()` both want their shapes to overlap some.
```openscad
module emboss(s=1000, overlap=0.001) {
    assert($children==2);
    difference() {
        children(0);
        down(overlap) bottom_half(s) children(1);
    }
    up(overlap) top_half(s) children(1);
}
```

closed 2020-12-17 by adrianVmariano