BelfrySCAD/BOSL2 #270: [BUG] Nested attachment `diff()`s work incorrectly.

revarbat, 2020-09-19:
**Describe the bug**
If you have nested attachment `diff()`, which are trying to use the same tag name, then the inner `diff()` works incorrectly.

**Code To Reproduce Bug**
```
module shape1() {
    diff("hole slot")
    cyl(d=50, h=20, $tags="body") {
        cuboid([10, 51, 21], $tags="slot");
        cyl(d=30, h=21, $tags="hole");
    }
}

diff("hole")
cuboid(100) {
    attach(TOP) shape1($tags="hole");
}
```

**Screenshots**
The expected correct result is:
[image]

The actual incorrect result is:
[image]

The shape of `shape1()` is:
[image]

revarbat, 2020-09-22:
Simple example:
```openscad module shape2() {
  diff("hole")
    cyl(d=50, h=20, $tags="body") {
    cyl(d=30, h=21, $tags="hole");
    }
}

diff("xhole")
  cuboid(50) 
    attach(TOP) shape2($tags="xhole");
It gets the wrong answer even without any repeated tags.

revarbat, 2020-11-12:
Fixed in the sense that you can now `diff()` an attached object that. is constructed with `diff()` itself.  However, the fix required makes it so that you can only use one `diff()`, `intersect()`, or `hulling()` on a parent object.

adrianVmariano, 2020-11-12:
Do you mean you can have only two levels of recursions of diff()?  (Or those other operations?)

revarbat, 2020-11-12:
You can do:
```openscad
diff("hole1")
shape1()
    diff("hole2")
    shape2()
        shape3($tags="hole2");
```
But not:
```openscad
diff("hole1")
diff("hole2")
shape1() {
    shape2($tags="hole1")
    shape3($tags="hole2");
}
```
or:
```openscad
diff("hole")
intersect("perim")
shape1() {
    shape2($tags="hole")
    shape3($tags="perim");
}
```

revarbat, 2020-11-12:
You can have 2 or more `diff()`, as long as no two are for the same parent object.  For any one parent object, you can have exactly one `diff()`, `intersect()`, or `hulling()`.

closed 2021-02-02 by revarbat