BelfrySCAD/BOSL2 #295: [BUG] cuboid rounding broken since speed optimizations

EarnyHerzberg, 2020-10-09:
4443be2 Merge pull request #286 from revarbat/revarbat_dev --> OK
63efb6f Merge pull request #287 from revarbat/revarbat_dev --> broken

Example:

include <BOSL2/std.scad>

cuboid(size = [15, 15, 4], rounding = 1, edges=[LEFT + FRONT, LEFT + BACK, RIGHT + FRONT, RIGHT + BACK]);
right(20)
cuboid(size = [15, 15, 4], rounding = 2, edges=[LEFT + FRONT, LEFT + BACK, RIGHT + FRONT, RIGHT + BACK]);
right(40)
cuboid(size = [15, 15, 4], rounding = 4, edges=[LEFT + FRONT, LEFT + BACK, RIGHT + FRONT, RIGHT + BACK]);
right(60)
cuboid(size = [15, 15, 4], rounding = 6, edges=[LEFT + FRONT, LEFT + BACK, RIGHT + FRONT, RIGHT + BACK]);

revarbat, 2020-10-09:
Oooh.  Okay, that's fixable.

revarbat, 2020-10-09:
Fixed in a53c77f

closed 2020-10-09 by revarbat

EarnyHerzberg, 2020-10-09:
Thanks! Great library...