BelfrySCAD/BOSL2 #307: [BUG] cuboid chamfer inconsistent at TOP / BOT

EarnyHerzberg, 2020-10-29:
Switch between

git checkout 4443be2  ## --> see Bug #295 also
git checkout master

and look at the middle two colums.

Example:

```
include <BOSL2/std.scad>

sz = [10,10,4];
cm1 = 1.9;
cm2 = 1.0;

cuboid(size = sz, chamfer = cm1, edges = [LEFT])
right(15)
cuboid(size = sz, chamfer = cm1, edges = [RIGHT])
right(15)
cuboid(size = sz, chamfer = cm1, edges = [TOP])
right(15)
cuboid(size = sz, chamfer = cm1, edges = [BOT])
right(15)
cuboid(size = sz, chamfer = cm1, edges = [FRONT])
right(15)
cuboid(size = sz, chamfer = cm1, edges = [BACK])
;
down(10)
cuboid(size = sz, chamfer = cm2, edges = [LEFT])
right(15)
cuboid(size = sz, chamfer = cm2, edges = [RIGHT])
right(15)
cuboid(size = sz, chamfer = cm2, edges = [TOP])
right(15)
cuboid(size = sz, chamfer = cm2, edges = [BOT])
right(15)
cuboid(size = sz, chamfer = cm2, edges = [FRONT])
right(15)
cuboid(size = sz, chamfer = cm2, edges = [BACK])
;
down(20)
cuboid(size = sz, chamfer = cm1, except_edges = [LEFT])
right(15)
cuboid(size = sz, chamfer = cm1, except_edges = [RIGHT])
right(15)
cuboid(size = sz, chamfer = cm1, except_edges = [TOP])
right(15)
cuboid(size = sz, chamfer = cm1, except_edges = [BOT])
right(15)
cuboid(size = sz, chamfer = cm1, except_edges = [FRONT])
right(15)
cuboid(size = sz, chamfer = cm1, except_edges = [BACK])
;
down(30)
cuboid(size = sz, chamfer = cm2, except_edges = [LEFT])
right(15)
cuboid(size = sz, chamfer = cm2, except_edges = [RIGHT])
right(15)
cuboid(size = sz, chamfer = cm2, except_edges = [TOP])
right(15)
cuboid(size = sz, chamfer = cm2, except_edges = [BOT])
right(15)
cuboid(size = sz, chamfer = cm2, except_edges = [FRONT])
right(15)
cuboid(size = sz, chamfer = cm2, except_edges = [BACK])
;

```

revarbat, 2020-10-29:
Turns out it's not a top-bottom inconsistency.  It's a bug at corners where only one edge is chamfered, and the edge chamfered is aligned with an axis which is shorter than twice the chamfer size.

revarbat, 2020-10-29:
Fixed in b3f2922

closed 2020-10-29 by revarbat

EarnyHerzberg, 2020-10-30:
Works now as expected. Thanks!