BelfrySCAD/BOSL2 #315: [BUG] cuboid neg. chamfer / rounding works only on TOP / BOT

EarnyHerzberg, 2020-11-17:
Example:  (only some of all combinations ;-)
```
include <BOSL2/std.scad>

n = -2;

ydistribute(25) {
    xdistribute(15) {
        cuboid(size = [10, 10, 10], chamfer = n, edges = [TOP] );
        cuboid(size = [10, 10, 10], chamfer = n, edges = [BOT] );
        cuboid(size = [10, 10, 10], chamfer = n, edges = [FRONT] );
        cuboid(size = [10, 10, 10], chamfer = n, edges = [BACK] );
        cuboid(size = [10, 10, 10], chamfer = n, edges = [LEFT] );
        cuboid(size = [10, 10, 10], chamfer = n, edges = [RIGHT] );
    }
    xdistribute(15) {
        cuboid(size = [10, 10, 10], chamfer = n, edges = [TOP + LEFT] );
        cuboid(size = [10, 10, 10], chamfer = n, edges = [BOT + LEFT] );
        cuboid(size = [10, 10, 10], chamfer = n, edges = [TOP + RIGHT] );
        cuboid(size = [10, 10, 10], chamfer = n, edges = [BOT + RIGHT] );
        cuboid(size = [10, 10, 10], chamfer = n, edges = [FRONT + LEFT] );
        cuboid(size = [10, 10, 10], chamfer = n, edges = [BACK + LEFT] );
        cuboid(size = [10, 10, 10], chamfer = n, edges = [FRONT + LEFT] );
        cuboid(size = [10, 10, 10], chamfer = n, edges = [FRONT + RIGHT] );
    }
}
```
If n is positive, everthing works as expected.

jwhevans, 2020-11-17:
What is the intended functionality? Inform the user that negative is not allowed or convert to the absolute value and accept the negative chamfer?

EarnyHerzberg, 2020-11-17:
Negative chamfer are described in the documentation --> https://github.com/revarbat/BOSL2/wiki/shapes.scad#cuboid

The examples uses TOP / BOT negative chamfers only. I've expected it works on the other sides also like it does on positive chamfer's.

jwhevans, 2020-11-17:
I never used this functionality. Thanks for the documentation link.

revarbat, 2020-11-17:
The intended use of negative chamfers is to use with `difference()` to chamfer holes in another object.

There is insufficient data in the `edges=` argument to specify which face of each edge to chamfer towards, so I just enabled TOP/BOTTOM. It could be viable, though, to allow "face-only" `edges=` values, where all edges for those faces is negatively chamfered.

So, it's currently working as intended, but it can be improved. Or at the least the documentation could use improvement.

adrianVmariano, 2020-12-17:
I think it would be nice to be able to do edges=FRONT or edges=[LEFT,RIGHT], if it's not too painful.  Can you support any non-confusing set of edges?  Like reject if three edges meet and accept otherwise?

lar3ry, 2020-12-23:
I was looking at the code above, and found the results rather confusing.
For example, the first cuboid has the TOP FRONT and TOP BACK chamfered.
Since it has edges = [TOP] specified, should it not have all four edges on the top chamfered.
Most of the rest of them baffle me as well, especially the last two in the front row and the last 4 in the back row.

adrianVmariano, 2020-12-23:
Your results are not the same as mine.  The edges = [TOP] case has four chamfers.  I think everything looks as expected if you remember that you can only chamfer top and bottom edges, so any attempt to chamfer other edges is just ignored.   The last four in the back row don't specify any top edges, so there are no chamfers.  

I notice that when you specify things like edges=FRONT that the chamfers are too long, like maybe it thinks it's going to chamfer all four edges.   That's definitely a bug.

lar3ry, 2020-12-23:
Interesting. Perhaps I need to download the latest BOSL2 again.
Not sure what you mean about edges = FRONT  making chamfers too long. They look fine to me, except that I would have thought it would have 4 chamfers on the front FRONT-TOP, -BOTTOM, -LEFT, -RIGHT.
[image]

adrianVmariano, 2020-12-23:
Wow, big-time bugs there.   They are almost all wrong, some of them very weirdly wrong.

adrianVmariano, 2020-12-23:
This bug manifests in RC3 but not in 2019.05.

adrianVmariano, 2020-12-23:
The current behavior is that for n>0 you can chamfer any set of edges you want to.  For n<0 you can only chamfer top or bottom edges.  This is because it's not clear how to have 3 chamfers meet at a corner.  My proposal is that any combination of negative chamfers (or roundings) should be permitted as long as 3 don't meet at a corner, otherwise assert an error.

lar3ry, 2020-12-23:
Getting more and more confused here.
In the docs for cuboid, I see an example:
cuboid([30,40,50], chamfer=5, edges=[TOP+FRONT,TOP+RIGHT,FRONT+RIGHT], $fn=24);

The picture showing what that does, shows chamfers on (what I consider to be) all thr right places. But when I run in ot RC3, I get this:
[image]

adrianVmariano, 2020-12-23:
[image]

Above is how the test looks with n>0 in 2019.05.   Like I said, it's badly broken in the RC.

adrianVmariano, 2020-12-23:
[image]

Negative chamfers look the same in both versions.  Above is what I get.  I object to the single edge chamfers extending past the edges of the cube.   And of course we observe that the four vertical edges never get chamfered, no matter what you specify.

adrianVmariano, 2020-12-23:
At some point Revar needs to switch over to running the examples in the RC and then we'll have to look through all the examples and probably find a lot more cases where things are broken.  The RC passes the regression tests, but they only test functions, and not all of them yet.

lar3ry, 2020-12-23:
Ahh. OK. I no longer have 2019.05
But I wonder, why can't three chamfers meet at a corner?
I mean, if all sides are chamfered, we have 8 places where three chamfers meet.

adrianVmariano, 2020-12-23:
Three *negative* chamfers can't meet at a corner because...which way do they all point.   Actually now that I think about it, a single negative chamfer in isolation is ambiguous about which way to point it.

lar3ry, 2020-12-23:
Oh, right. This is the first time I realized that we don't seem to be able to put negative chamfers on the top, with the slopes rising at 45 degrees upward (if that makes any sense).

adrianVmariano, 2020-12-23:
Seems like it's complicated to decide which way to point the chamfers.  What if somebody requests negative chamfers and sets edges="Z".  What should happen?  It's totally ambiguous.

adrianVmariano, 2020-12-23:
Right.  Do you think that's limiting?  How could the cuboid interface support controlling the direction of chamfers?

lar3ry, 2020-12-23:
I don't know if it's even useful, but perhaps something like 
edges = [TOP], direction = [UP]
edges = [RIGHT], direction = [FRONT]
?

lar3ry, 2020-12-23:
Or even
edges = [TOP_right]
edges = [RIGHT+front]
?

adrianVmariano, 2020-12-23:
I think we should stick to the convention that predefined vars are all caps.  At least for normal variables.  (Probably not functions.)

Is the direction suggestion that entries in the lists are matched piece by piece, so edges=[TOP,BOTTOM], direction=[UP,DOWN] would produce the standard result?

adrianVmariano, 2020-12-24:
Actually would direction=UP cause the edges to point up, extending the side faces?  (Not what currently happens?)  And if so, what direction to the "normal" chamfers point?  It seems kind of like they are "out".  

Another idea would be to define canonical chamfer directions as is already done, where the left and right chamfers point out to the left and right respectively,  You'd have to rotate the cube if you wanted upward pointing chamfers.  And some cases might be impossible....

revarbat, 2020-12-24:
We have an interface for `edges()`, and one for `corners()`.  Perhaps we need a `faces()` interface that cuboid can use for negative chamfers with a `faces=[TOP,FRONT]` sort of arg.

revarbat, 2020-12-24:
I'm thinking something like:
```openscad
    cuboid(100, chamfer=-15, faces=[TOP,FRONT], except_edges=TOP+FRONT);
```

revarbat, 2020-12-24:
Regarding the weird chamfering in RC3, I'm wondering if this is related to the tweaks to boolean ops handling of empty children.

revarbat, 2020-12-24:
Verified bug in OpenSCAD RC3.  Filed issue openscad/openscad#3547

Minimal code to reproduce:
```openscad
hull() {
  translate([-4,-1,1]) cube(2, center=true);
  translate([3,0,0]) {
    intersection() {
      rotate([90,0,0]) {
        linear_extrude(height=4, center=true) {
          circle(r=2, $fn=4);
        }
      }
      translate([0,0,0]) {
        translate([0,0,0]) {
          mirror([0,0,1]) {
            translate([0,0,0]) {
              linear_extrude(height=4, center=true, convexity=2, $fn=4) {
                circle(r=2, $fn=4);
              }
            }
          }
        }
      }
    }
  }
}
```

adrianVmariano, 2020-12-24:
Weird that the result is OS dependent.

adrianVmariano, 2020-12-25:
> I'm thinking something like:
> 
> ```scad
>     cuboid(100, chamfer=-15, faces=[TOP,FRONT], except_edges=TOP+FRONT);
> ```

So what does this do?  Does it "extend" the top face and the front face?  How would you specify the other way of applying negative chamfers to the top face?  Would you need something like faces=[LEFT,RIGHT,FRONT,BACK] except_edges=[BOTTOM,"X"] ?  That seems sort of awkward.  

Maybe I don't understand.   I am also not sure of the advantage of introducing the new faces keyword.  Would faces and edges be used together somehow?

I was imagining maybe a scheme where you specify edges and a default direction is chosen that maintains consistency across corners, and then combine that with a flip keyword that flips edges (and all the connected ones to maintain consistency).   Seems kind of complicated, though.

adrianVmariano, 2020-12-25:
[image]
Rounded prism can do the reverse form of rounding.  But only on the top and bottom.  

    rounded_prism(square(10), height=10, joint_top=[2,-2], joint_bot=0, joint_sides=[0,0,0,0]);

revarbat, 2021-01-02:
I fixed the negative chamfer/rounding overhangs issue in commit a683fa9ead6ca5017e472456123bea50cf50b23a

revarbat, 2021-01-02:
The bad triangular chamfers issue is a bug in OpenSCAD and is being addressed by PR openscad/openscad#3568

revarbat, 2021-01-02:
The interface issue for how to specify non-TOP/BOTTOM negative chamfers/roundings is all that remains to be addressed.

adrianVmariano, 2021-01-02:
What are your thoughts on the ideas proposed thus far for specifying non TOP/BOTTOM negative chamfers.

revarbat, 2021-01-02:
The current TOP/BOTTOM edges for negative chamfers design has the benefits of being easy to understand and consistent, as long as it's documented clearly.
If I want FRONT/BACK negative chamfers, it's trivial to use `orient=BACK`.  For LEFT/RIGHT, `orient=RIGHT`.
If, for some weird reason, I want FRONT and TOP negative chamfers, I can do it with a second cuboid.

The interface solutions to this all seem more confusing than just rotating or having a second cuboid.

I think, perhaps that it SHOULD assert an error if you try to do a negative chamber of the Z aligned edges. (Unless edges=EDGES_ALL)

adrianVmariano, 2021-01-02:
Ok.  What about reversed negative chamfers?

adrianVmariano, 2021-01-05:
Is this resolved?   What still needs to be done?

revarbat, 2021-01-05:
Still need to add asserts against using Z aligned edges with negative chamfers/roundings.

adrianVmariano, 2021-01-08:
Hull bug fixed in RC5.

revarbat, 2021-05-08:
Asserts added in 3f25aa7d8797186ae3928eb594190845e3adb559