BelfrySCAD/BOSL2 #318: [BUG] cuboid with a rounding = 0 fails to draw if $fn is > 96

DaleSchultz, 2020-11-29:
**Describe the bug**
a cuboid with a rounding = 0 fails to draw if $fn is > 96

**Code To Reproduce Bug**
```
include <BOSL2/std.scad>
// cuboid rounding = 0 with $fn > 96 fails to draw
cuboid(p1=[ 0,0,0], size=[5,20,8], rounding = 0    ,$fn=96); // draws fine 
cuboid(p1=[10,0,0], size=[5,20,8], rounding = 0    ,$fn=97); // does not draw
cuboid(p1=[20,0,0], size=[5,20,8], rounding = 0.001,$fn=97); // draws fine
```

**Expected behavior**
Expect it to draw and render with $fn > 96

**Screenshots**
[image]

**Additional context**
a rounding value of .001 makes it work...

- OpenSCAD Version: 2019.05

- Other libraries used: only BOSL2/std.scad

revarbat, 2020-11-30:
I cannot seem to reproduce this bug with either OpenSCAD 2019.05 or 2020.09.26

DaleSchultz, 2020-11-30:
wow, what other settings are possible I wonder....
I spent hours narrowing it down to that combination. Let me restart OpenSCAD in case something is stuck there....

DaleSchultz, 2020-11-30:
yup, after restarting OpenSCAD, it works perfectly!
OK I learned to do that too now when seeing strange things.
I wonder how I managed to get it into that state....

Really appreciate the whole library - many thanks, great work!

closed 2020-11-30 by DaleSchultz

DaleSchultz, 2020-12-01:
it is very similar in naure to

revarbat, 2020-12-02:
I still cannot reproduce this.

revarbat, 2020-12-02:
I did change `cuboid()` to specifically check if rounding=0 or chamfer=0 and just give `cube()`.