BelfrySCAD/BOSL2 #321: ignore edges= if rounding = 0

DaleSchultz, 2020-12-01:
If one defines the edges to be rounded on a cuboid, and one has the rounding value customizable, setting the rounding=0 results in the cuboid not being drawn, with no error warning.

If rounding is zero, the edges should be ignored.

**Example Code**
```
include <BOSL2/std.scad>
// cuboid rounding = 0 with edge fails to draw
cuboid(p1=[ 0,0,0], size=[5,20,8], rounding = 0, edges=TOP); // does not draw
cuboid(p1=[10,0,0], size=[5,20,8], rounding = 0, edges=[]); // does not draw
cuboid(p1=[20,0,0], size=[5,20,8], rounding = .1, edges=TOP); // draws fine
```

revarbat, 2020-12-02:
I cannot reproduce this in OpenSCAD 2019.05 or 2020.09.26

revarbat, 2020-12-02:
May be a duplicate of #318

DaleSchultz, 2020-12-02:
well #318 went away after restarted OpenSCAD, but this one does not.
I just started OpenSCAD fresh and ran those lines.
[image]
I am running on Windows 64bit. What else can I specify to help reproduce?

I made this one an 'enhancement" because technically one could argue that defining edges makes no sense if rounding = 0

I trust you mean that if rounding = zero you revert to translate() cube()....

Thanks again.

revarbat, 2020-12-02:
I'm on Mac OS Catalina, 64 bit.
You sure you have the latest BOSL2 repository?
[image]

DaleSchultz, 2020-12-02:
I downloaded it on November 18
echo(bosl_version_str());  // 2.0.469

revarbat, 2020-12-02:
Try getting the latest, and try again. I added a few tweaks yesterday that should make this bug impossible,

DaleSchultz, 2020-12-02:
indeed version 2.0.473 draws all three!
Thanks.

Do you think something with the Windows environment could be making the difference?

I also tried -0

revarbat, 2020-12-03:
Unclear, really.  I don't have a Windows environment to test with at the moment.

revarbat, 2020-12-03:
This specific bug is fixed in latest BOSL2, so I'll close this.

closed 2020-12-03 by revarbat