BelfrySCAD/BOSL2 #323: [BUG] Trapezohedron - problems with assert parmcount==2

charmaur, 2020-12-02:
**Describe the bug**
Trapezohedron fails (parmcount == 2) assert when using all combinations of (faces, r, side, longside, h, d) except for side and r.
```
// side and r = success
regular_polyhedron("trapezohedron",faces=10, side=1, r=1.45, facedown=false);

// the following all produce this error
// ERROR: Assertion '(parmcount == 2)' failed: "Must define exactly two of 'r', 'side', 'longside', and 'height'" in file BOSL2/polyhedra.scad, line 737

// side and longside
regular_polyhedron("trapezohedron",faces=10, side=1, longside=2.25, facedown=false);

// h and r
regular_polyhedron("trapezohedron",faces=10, h=2.4, r=1.45, facedown=false);

// side and h
regular_polyhedron("trapezohedron",faces=10, side=1, h=2.4, facedown=false);

// longside and h
regular_polyhedron("trapezohedron",faces=10, longside=2.25, h=2.4, facedown=false);

// longside and r
regular_polyhedron("trapezohedron",faces=10, longside=2.25, r=1.5, facedown=false);
```

**Code To Reproduce Bug**
```
include <BOSL2/std.scad>
include <BOSL2/polyhedra.scad>
// from the example
regular_polyhedron("trapezohedron",faces=10, side=1, longside=2.25, facedown=false);
```

**Expected behavior**
Ability to specify the following combinations for trapezohedron parameters:
- side and longside
- h and r
- side and h
- longside and h
- longside and r

**Additional context**
Is the dflt=1 in function _trapezohedron let "r = get_radius(r=r, d=d, dflt=1)" setting r even when not specified?

- OpenSCAD Version: v2.0.402 and v2.0.473

Love the library. Thank you!

adrianVmariano, 2020-12-03:
This should be fixed in the latest release.  Can you confirm that everything works?

charmaur, 2020-12-03:
Working perfectly. Thank you!

closed 2020-12-03 by charmaur