BelfrySCAD/BOSL2 #335: [BUG] Tests should be repeatable

thehans, 2020-12-14:
**Describe the bug**
While looking into the OpenSCAD crash reported on mailing list, I noticed that `tests/hull.scad` uses `rands` without a seed, so results are not repeatable between runs.

**Code To Reproduce Bug**
preview `tests/hull.scad` more than once

**Expected behavior**
Any tests involving (pseudo)random data should use a seed.  I've only looked at hull.scad but maybe others also have similar issues.

I ended up editing my copy of tests/hull.scad with the following:
```OpenSCAD
seed = 2;
rnd1 = rands(-1,1,21,seed);
rnd2 = rands(-1,1,21,rnd1[20]);
rnd3 = rands(-1,1,21,rnd2[20]);
rnd4 = rands(-1,1,21,rnd3[20]);
rnd5 = rands(-1,1,151,rnd4[20]);

testpoints_coplanar = let(u = unit([1,3,7]), v = unit([-2,1,-2])) [ for(i = [1:10]) rnd1[i*2-2] * u + rnd1[i*2-1] * v ];

testpoints_collinear_2d = let(u = unit([5,3])) [ for(i = [1:20]) rnd2[i-1] * u ];
testpoints_collinear_3d = let(u = unit([5,3,-5])) [ for(i = [1:20]) rnd3[i-1] * u ];

testpoints2d = 20 * [for (i = [1:10]) [rnd4[i*2-2], rnd4[i*2-1]] ];
testpoints3d = 20 * [for (i = [1:50]) [rnd5[i*3-3], rnd5[i*3-2], rnd5[i*3-1]] ];
```
I chose to add an extra value to each rands call. and chain the seeds based on that, but they could be chained in other ways as long as some seed is used each time.

Since the range of values is always the same in this case, a single call to `rands` could also be used, with enough elements to fill all testpoints* with unique random values, by starting each at different offsets.  Or even just re-use the same random values, starting at index 0 for all testpoints vectors, since the values are applied differently anyways.

adrianVmariano, 2020-12-15:
Definitely agree with this.  The crash problem I observed was with test_hull.scad, not with hull.scad, and I generated random examples but then hard coded them so there is nothing random.  Actually do we need hull.scad any more at all?  What does it do that test_hull.scad does not?  (Or maybe it should be put somewhere else as an example instead of a test?)

adrianVmariano, 2020-12-15:
Actually tests/hull.scad dumps core for me too.  So does tests/polyhedra.scad.

thehans, 2020-12-16:
Sorry for my mixup of the filenames.

After trying out `scripts/run_tests.sh` (all passing with https://github.com/openscad/openscad/pull/3542) I see that it doesn't even use `tests/hull.scad` (only files matching `test_*`).  So yeah I'm not familiar with this project enough to know if it still has any purpose.  Was hull.scad originally maybe used to generate the hardcoded values in tests_hull.scad or something?

Likewise `polyhedra.scad` doesn't begin with `test_` so it is not run by `srcipts/run_tests.sh` either.  So its not clear what its purpose is.

Back on the topic of rands, i did a quick grep of usages and found these (besides already mentioned tests/hull.scad )
 - `tests/test_geometry.scad` a lot of uses without seed.  some later in the file do specify `seed_value` (should be just `seed`, but openscad picks it up as a positional parameter anyways)
 - `tests/test_math.scad` a couple usages here
 - `/arrays.scad` rands used in `function shuffle(list)`, maybe good idea to add optional seed parameter on shuffle to pass down if provided.  Didn't check what sort of test cases are done on shuffle.

adrianVmariano, 2020-12-16:
The presence of hull.scad and polyhedra.scad are kind of historical, I think.  They are both really examples rather than tests, but when they were included in the tests directory, no tests existed for either.  The hull.scad was the example set that originally came with the convex hull code.  And polyhedra.scad demonstrates almost all of the polyhedra.  But we have no mechanism for checking validity of geometry outputs, so no way to actually use polyhedra.scad as a test.  I suppose I need to run through all possible polyhedra and code checks that the right vertices (and faces?) come out.  

The tests for shuffle look kind of suspect.  The test shuffles an array twice and checks that the resulting three arrays are all different.  So you could be "lucky" and get a match and then you'd fail the test.   (Admitedly pretty unlikely with 100 items in the list.)   It seems like the only truly valid test would be to check that the output has all the entries the input had.  I guess if we assume that the algorithm doesn't change we could code for a specific shuffled result.

revarbat, 2020-12-17:
The existence of tests/hull.scad and tests/polyhedra.scad are, for the most part, placeholders for yet-to-be-implemented tests for the `module`s in those files.  They are usable as manually-verified tests at the moment.

adrianVmariano, 2020-12-17:
test/test_hull.scad fully tests the contents of hull.scad---except for the "fast" module which isn't even in tests/hull.scad anyway.    So there's no need to hold a place for anything.

thehans, 2020-12-17:
> But we have no mechanism for checking validity of geometry outputs, so no way to actually use polyhedra.scad as a test. I suppose I need to run through all possible polyhedra and code checks that the right vertices (and faces?) come out.

You could crib from OpenSCAD's own test suite if you were so inclined.  It uses ctest framework, which calls to some python scripts that handle invoking openscad, diffing .echo results, and comparing image exports against expected images using imagemagick.  It may all be a bit much for a library test, but I think it serves its purpose pretty well on OpenSCAD side.

revarbat, 2020-12-17:
> test/test_hull.scad fully tests the contents of hull.scad---except for the "fast" module which isn't even in tests/hull.scad anyway. So there's no need to hold a place for anything.

Nuke it then.

adrianVmariano, 2020-12-22:
I added seeds to all the tests and  a seed parameter to shuffle().   See PR.

What is the correct parameter name for the rands() seed parameter?  The manual says it's seed_value?  Is it wrong and it's actually just seed?

thehans, 2020-12-22:
Its `seed`, going by the autocompletion suggestion.  However, I guess its all a moot point, because after looking at the source and testing a bit, `rands` doesn't pay attention to parameter names, *only* their position.

revarbat, 2020-12-23:
I concur. Built-in functions only care about argument position.

adrianVmariano, 2020-12-23:
I didn't realize that.

Do you want to save the suggestion above for how to do regression testing on modules before we close this issue?

revarbat, 2020-12-23:
I already have a scheme in mind for testing modules that will be more accurate than comparing generated PNGs.
I just have to write the script for it.

adrianVmariano, 2020-12-23:
Ok.  Then the PR fixes this bug.

closed 2020-12-23 by adrianVmariano