BelfrySCAD/BOSL2 #342: cylinder odd result with only r1 given

adrianVmariano, 2020-12-19:
If you make a cylinder by just giving r1 a default r2=1 is used.  This is both weird and doesn't match base OpenSCAD, which sets r2=r1 in this case.  I wonder if there are other cases of this kind of weirdness when there are two radii?  Or other ways in which the revised primitives disagree with base openscad.

adrianVmariano, 2020-12-19:
cylinder(h=5,r=5);
right(7)
cylinder(h=5,5);
right(14)
cylinder(5,5);
right(22)
cylinder(5,r2=5);
right(32)
cylinder(5,r=5);
Seems defaults are coming into play.
But in unexpected ways

lar3ry, 2020-12-19:
Just discovered that the same thing happens in 2019-05 without using BOSL2.

cylinder(5,3);
[image]

adrianVmariano, 2020-12-20:
I'm confused.  Same as what?  Is that a cone with radii 3 and 1 and height 5?  

The basic problem is that we need for cylinder to behave exactly the same in BOSL2 as out of it so that code doesn't break because you're in BOSL2.

lar3ry, 2020-12-20:
Exactly. That was the result in 2019.05
I was not in BOSL2.

revarbat, 2020-12-20:
I get the same exact results in OpenSCAD 2020.12.11 whether or not BOSL2 is included, using the following code:

revarbat, 2020-12-20:
I cannot reproduce this bug.

closed 2020-12-20 by adrianVmariano