BelfrySCAD/BOSL2 #346: [BUG] BOSL2 sphere may be very odd

RonaldoCMP, 2020-12-23:
BOSL2 sphere overrides the buit-in OpenSCAD sphere. But even with the default parameters they show quite radical differences:

[image]
Built-in and BOLS2 default spheres

The main reason for that difference is they have different criteria to distribute the polyhedron vertices on the sphere surface. I will be back to this point later.

For radius values less than 2e-4, the differences are even more dramatic:

[image]

For some radius values, the BOSL2 sphere vertices may not be on the sphere surface.

I have not read the BOSL2 sphere code to check it, but I suspect that the vertices are correctly computed. The issue is that polyhedron may be discarding/altering vertex count and coordinates for small radius spheres. For instance, scaling the radius up and the sphere down by the same factor works as expected. To circumvent that it seems to be enough call:

    scale(r) sphere(1,$fn);

instead of 

    sphere(r,$fn);

NOPLib has a module Sphere() that builds a polyhedron. Its main goal is to assure the sphere bounding box is a cube with edge length equal to the sphere diameter. It approximates $fn by a value multiple to 4 in order to have vertices on the positive and negative rays of the 3 coordinate axis. I think that if BOLS2 sphere geometry does not agree with the builtin sphere geometry than the NOPLib strategy is preferable. Anyway, the two alternatives might be user selected by an appropriate parameter of BOSL2 sphere.

adrianVmariano, 2020-12-23:
Note that the sphere module has a "style" parameter that can select different types of sphere.  I think by default it should be the same as the native sphere, though.   The manual is kind of vague about the properties and specific characteristics of the different styles.

revarbat, 2020-12-24:
`sphere()` and `spheroid()` can be given `style=` arguments where the value can be one of "orig", "aligned", "stagger", "octa", or "icosa":

- `style="orig"` is supposed to construct a sphere the same way that the OpenSCAD sphere built-in does.  This is the default for `sphere()`.
- `style="aligned" constructs a sphere where, if `$fn` is a multiple of 4, it has vertices at all axis maxima and minima.  ie: it's bounding box is exactly the sphere diameter in length on all three axes.  This is the default for `spheroid()`.
- `style="stagger"` forms a sphere where all faces are triangular, but the top and bottom poles have thinner triangles.
- `style="octa"` forms a sphere by subdividing an octahedron (8-sided platonic solid).  This makes more uniform face areas over the entirety of the sphere, and guarantees the bounding box is the sphere diameter in size on all axes.  The effective `$fn` value is quantized to a multiple of 4, though.
- `style="icosa"` forms a sphere by subdividing an icosahedron (20-sided platonic solid).  This makes even more uniform face areas over the entirety of the sphere.  The effective `$fn` value is quantized to a multiple of 5, though.

Now, if the overridden `sphere()` is not making an "orig" style sphere, that's definitely a bug.

RonaldoCMP, 2020-12-24:
It seems there is a bug in rotate_extrude. The code:

sides=12; r = 1e-4;
rotate_extrude(convexity=2,$fn=sides) {
    n = floor(sides/2);
    scale(1000)
    polygon([for(i=[0:n]) r*[cos(-90+i*(180/n)), sin(-90+i*(180/n))] ]);
 }

produces complete wrong results.

RonaldoCMP, 2020-12-24:
The issue of rotate_extrude seems to be that it internally calls polygon(). The primitive polygon() has the bad habit of eliminating polygon vertices that are near to each other. 

The spheroid() code is basically:
```
sides=12; r = 1e-4; 
rotate_extrude(convexity=2,$fn=sides) { 
    scale(1000) 
    difference() { 
        circle(r=r,$fn=sides);
        translate([-r,0,0]) square(2*r,center=true);
    }
 }
```
If we render the difference it is fine but the object is bogus possibly because rotate_extrude called something like polygon() to render the difference.

adrianVmariano, 2020-12-24:
We need a quantization table that lists how the various primitives quantize.  I know that when we wrote the ruler module we ran into this problem and we couldn't make a ruler finer than about 1e-3.

revarbat, 2020-12-25:
The shapes in shapes2d.scad are quantized. `square()` and `circle()` are not overridden, so they are not quantized. `cube()`, `cuboid()`, `cylinder()`, `cyl()`, `sphere()` and `spheroid()` are not quantized.

revarbat, 2020-12-25:
Basically, only 2D shapes that use `polygon()` to form are quantized. However any 2D union/difference/intersection/offset/minkowski operation in I'll quantize a shape.

adrianVmariano, 2020-12-25:
Everything is quantized, isn't it?  Just a question of what the quantum is.

revarbat, 2020-12-25:
All 2D ops scale all vertices by 2048, and converted them to integers to pass to ClipperLib, before scaling them back by 1/2048.  Put another way, 2D ops quantize to 2^-11
As best as I can tell from the code, `polyhedron()` quantizes to 1/1048576 or 1/(1024*1024) or 2^-20.

revarbat, 2020-12-25:
Okay, so `sphere()` calls `spheroid()`, which attempts to make a fast sphere via `rotate_extrude()` of a half-circle.
Problem: Making that half circle requires a `difference()` which quantizes the vertices.  Making the half-circle via `polygon()` may also quantize the vertices, I think.

If I remove the `rotate_extrude()` optimizations, then my code generates a sphere via `polyhedron()` that is correct and isn't so badly quantized.  However it's slower at making spheres.

adrianVmariano, 2020-12-25:
Why is making a sphere via polyhedron slow?  Usually polyhedron is fast.  Is it the code to compute the faces?

revarbat, 2020-12-25:
Honestly, I haven't benchmarked it.  I just cannot imagine calculating all vertices of the sphere being faster than using the builtins to calculate them.

adrianVmariano, 2020-12-25:
Well, it's not going to be faster, clearly, but it's not obvious that it is necessarily significantly slower.

revarbat, 2020-12-25:
10,000 spheres created via built-in `sphere()`: 0.53 seconds.
10,000 spheres created via `polyhedron()`: 19.65 seconds.

So 37x slower.

revarbat, 2020-12-25:
10,000 spheres created via `rotate_extrude() polygon(half_circle_path);`: 8.725 seconds.

revarbat, 2020-12-25:
Interestingly, 10,000 spheres created via `rotate_extrude()` of a circle with half of it differenced away takes 14.556 seconds.

adrianVmariano, 2020-12-25:
What's the largest number of spheres in a reasonable model that renders in, say, less than an hour?

adrianVmariano, 2020-12-25:
Is the time for computing the vertices, computing the faces, or actually making the polyhedron?   Like if you compute the vertices and faces once and then reuse them 10k times, what is the time to use polyhedron?

revarbat, 2020-12-25:
Well, the 10000 spheres I was rendering were all r=1, $fn=5, in a 100x100 grid.  You would have to precompute or cache spheres for each $fn value, so that may be unviable.

revarbat, 2020-12-25:
If I comment out the `vnf_polyhedron()` call, the time halves, but I suspect most of that is in OpenSCAD unioning the spheres.

RonaldoCMP, 2020-12-25:
As far as I could check, the following is equivalent to the builtin sphere(r=r,$fn=$fn):

```
  sides = r<1/pow(2,20)? 3: $fn;
  circum  = false;
  mirror([1,0,0])
  rotate_extrude(convexity=2,$fn=sides) { 
      scale(r) 
      difference() { 
          oval(r=1, circum=circum, realign=(sides%4!=1), $fn=2*ceil(sides/2));
          left(1) square(2,center=true);
      }
   }
```

To be consistent with that, other styles may require some tweak.

adrianVmariano, 2020-12-25:
Is there any point in trying to make things work at tiny scale, where OpenSCAD is unreliable and things go crazy anyway?  If you want to make something really small you probably need to make it big and then scale it down, because you're likely to run into all sorts of problems.

adrianVmariano, 2020-12-25:
The time for preview doesn't include any union.  I timed render on sets of nonintersecting spheres, 10 spheres took 3s, 100 took 1minute, 200 took about 2 minutes, 400 took 5 minutes.  More linear than I was expecting.

revarbat, 2020-12-28:
I've made the BOSL2 "orig" sphere match the built-in sphere as exactly as possible for both default size and r=2e-4.
committed in 53fae5a3b29c66431459deb1c6d2144cf291560c

RonaldoCMP, 2020-12-28:
The default $fn for spheres with style="aligned" has an odd value for radius less that 4; so, it is not aligned at all. It should be a multiple of 4. Besides, for radius less than 3, the sphere radius is less than the specified. I think that this style should redefine the user's $fn as a multiple of 4 to make sense.

revarbat, 2020-12-29:
A spheroid with style "aligned" is documented as only having maxima and minima at the sphere radius when $fn is divisible by 4.  That doesn't *require* $fn to be divisible by 4, though.  In fact an "aligned" spheroid of arbitrary $fn is useful as an alternate shape to the "orig" sphere triangulation.

revarbat, 2020-12-29:
As far as I can tell, size of "aligned" spheroid with r=3 is correct.

closed 2020-12-29 by RonaldoCMP