BelfrySCAD/BOSL2 #347: Find the bezier point(s) for given x, y or line

martincizek, 2020-12-27:
I repeatedly came across the need for getting the bezier curve point for a given `x` or `y`. It was related to use cases where the bezier forms the top surface of an object that should hide some fixed-height features inside. The existing function `bezier_segment_closest_point()` was not appropriate for it.

I developed a workaround using the bisection method, see below. It is definitely not as elegant as the mentioned `bezier_segment_closest_point()`.

So what I'd appreciate is some more support for locating exact point(s) on the bezier curves and paths, not only points closest to a given point. I believe it might be even simpler to implement than `bezier_*_closest_point()` functions.

Does it make sense? Or can this be achieved in other ways? I guess the upcoming lambda functions in OpenSCAD might be useful for similar APIs(?)

This is the workaround I am using now:
``` openscad
BEZIER_U_TOLERANCE = 1e-9;

// Function: bezier_segment_u_at_y()
// Usage:
//   bezier_segment_u_at_y(bezier, y, [u1], [u2])
// Description:
//   Finds the part of the given bezier segment where the curve crosses `y`.
//   The degree of the curve, N, is one less than the number of points in `curve`.
//   Returns [`u`, `valid`] where the bezier segment matches `y` (`valid` flag
//   is true) or where the bezier segment is closest to the `y` (`valid` flag
//   is false).
//   Limitations: If there are more crossings, just one is picked. You can influence
//   this by restricting the segment part range by setting `u1` and/or `u2` parameters.
// Arguments:
//   curve = The list of endpoints and control points for this bezier segment.
//   y = The y value to find the matching curve point.
//   u1 = the minimum `u` to search the crossing in.
//   u2 = the maximum `u` to search the crossing in.
function bezier_segment_u_at_y(curve, y, u1=0, u2=1) =
  let(
    um = (u1 + u2) / 2,
    dy_u1 = bezier_points(curve, u1).y - y,
    dy_um = bezier_points(curve, um).y - y,
    go_right = (dy_u1 >= 0) == (dy_um >= 0),
    new_u1 = go_right ? um : u1,
    new_u2 = go_right ? u2 : um,
    stop = (u2 - u1) < BEZIER_U_TOLERANCE,
    valid = stop && abs(dy_um * (um - u1) / (dy_um - dy_u1)) < BEZIER_U_TOLERANCE
  )
  stop ? [um, valid] : bezier_segment_u_at_y(curve, y, new_u1, new_u2);
```

adrianVmariano, 2021-01-02:
So you're saying you have a bezier curve, p(t)=[x(t),y(t)] and you want to find the points y where p(t) = [x0,y] for some x0?  Is that right?  In principle this could be done by calling the poly_roots function for the bezier polynomial, but that does require determining the coefficients of that polynomial.   

Why do you want these crossings?  What do you do with them?

RonaldoCMP, 2021-01-12:
The algebraic coefficients of a Bezier curve may be computed by a non-public matrix function defined in `beziers,scad`. The task asked for by martincizek may be accomplished by a 4 line code:

```
// Function: bezier_line_intersection()
// Usages: 
//   params = bezier_line_intersection(p, line);
// Description:
//   Finds the parameters of the 2d curve whose Bezier control points are `p`
//   corresponding to its intersection points with the line through
//   the pair of distinct 2d points in `line`. The returned parameters, if some, are
//   not necessarily in the interval [0,1].
// Arguments:
//   p - list of 2d control points of the Bezier curve
//   line - a list of two distinct 2d points defining a line
function bezier_line_intersection(p, line) =
    assert(is_path(p) && len(p[0])==2, "The input ´p´ must be a 2d path")
    assert(_valid_line(line,2), "The input `line` is not a valid 2d line")
    let( 
        a = _bezier_matrix(len(p)-1)*p, // curve algebraic coeffs. 
        n = [-line[1].y+line[0].y, line[1].x-line[0].x], // line normal
        q = [for(i=[len(a)-1:-1:1]) a[i]*n, (a[0]-line[0])*n] // curve SDF to line
    )
    real_roots(q);
```

To get the intersection points it is enough to evaluate the curve at the returned parameter values with `bezier_points()`.

martincizek, 2021-02-01:
> Why do you want these crossings? What do you do with them?

@adrianVmariano I'm generating an optimum smooth path from one floor surface to another floor surface - futuristic replacement of doorsteps. :) The basic solution is to smooth a two-point path with horizontal tangents, see below.

Smooth-curvature ramp from one floor to another:
[image]

The constraint is that there is a cliff (in red) that needs to be hidden under the curve. If I were lucky, the cliff would just fit. But this is how the unlucky case looks like:
[image]

So my approach is to take the midpoint's position + tangent and extend the path to the left, so that the final curve touches the cliff. The final curve is then defined by `smooth_path()` of three points (1st - horizontal tangent, 2nd - midpoint's tangent, 3rd - horizontal tangent). The result looks like this:
[image]

This approach fulfils my requirements:
* optimum space usage - not adding more X-size than required
* curve's curvature is changing smoothly
* the curve is changing continuously with the cliff size - i.e. if the cliff interferes the "happy path" just a little, the resulting curve looks almost the same as if the cliff was just below the path.

And I didn't find any other solution even with higher-degree beziers and the approach of "smooth" method in `round_corners()`.

If there is a more elegant solution, would appreciate it. :)

martincizek, 2021-02-01:
> The algebraic coefficients of a Bezier curve may be computed by a non-public matrix function defined in `beziers,scad`. The task asked for by martincizek may be accomplished by a 4 line code:

@RonaldoCMP Wow, that's look way more elegant than my bisection method. :) Thanks!

Anyway, it seems that function literals are just being introduced to BOSL2 and I guess such feature(s) should come along similar features for e.g. normal paths. So leaving it up to the maintainers what to do with this ticket. :) Although my brute-force approach works and Ronaldo's non-public function approach would probably work too, function literals can provide a nice way how to address find-something tasks.

P.S. the library is just fantastic, thanks for your great work!

RonaldoCMP, 2021-02-01:
I am afraid that function `smoth_path()` will not fulfill your requirement of continuous curvature: it is only tangent continuous.

martincizek, 2021-02-01:
> I am afraid that function `smoth_path()` will not fulfill your requirement of continuous curvature: its solution is only tangent continuous.

I see. I saw some discussion comparing beziers to clothoid / track transition curve, concluding that even cubic bezier can be a reasonable approximation (at least aesthetically, IIRC). And the above curve looks OK-ish to me. Practically speaking, this should a) look good when printed and b) make robot vacuum reasonably fluent when overcoming the height difference :).

Given that this is hopefully still way better than a linear wedge (and an engineer-level solution is OK here), do you think there is an easy-to-construct alternative curve that would do the job noticeably better?

adrianVmariano, 2021-02-01:
You can construct a continuous curvature curve using 4th order beziers, but you'll have to set up the control points yourself.   If you define the points with zero curvature as the start, finish, and "corner" point, then you need 5 control points to define each of two beziers (9 distinct points total).  The construction is straight forward, but you have 4 degrees of freedom to address (like the k factor in round_corners).  And for this simple construction to work there has to be enough space so that the tangent line of the start point and the midpoint intersection between each other and not before the start of your curve.  

I question the importance of continuous curvature for  your application.  I think it may produce a path with a higher slope that is less friendly to robot vacuums.   Of course, it may also be the case that the robot vacuum prefers the ramp, because it's a simpler control situation: abrupt change in slope, then constant.  

We have not yet added function literals to BOSL2 in any meaningful way.  What is your idea for how we might exploit this new feature?  I didn't understand what you were thinking.

martincizek, 2021-02-01:
> You can construct a continuous curvature curve using 4th order beziers, but you'll have to set up the control points yourself. [...] but you have 4 degrees of freedom to address (like the k factor in round_corners).

That I was trying to do ... and as you say, there was too much freedom. And I've concluded it would be inadequate effort to figure out appropriate points that can be automatically calculated based on a few parameters (which would be height, cliff point and one or two deviance parameters).

> I question the importance of continuous curvature for your application. 
You're right. That was incorrectly named requirement. The true requirement is just friendliness for robovac movement. :) The robovac is around 30 cm long, the ramp around 20 cm (orthogonally - limit of the printer) and the hight to overcome only 3.5 cm, so I hope any choice would work. But 12 cm of this 20 cm is the cliff, so I was hoping to use the space above cliff in best possible way for the curve.

> We have not yet added function literals to BOSL2 in any meaningful way.
I've noticed it in `heightfield()`.

In this case, I was thinking of using it as a generic wrapper around any planar function (like the line in this case) to find its minimum/maximum/given value along the path, bezier function, etc. But I was probably too closed in my iterative solution finding mindset (as I started with the bisection method).

But to explain myself, it might look like:
``` scad
// Specialized case (optimized with analytic solution):
line_closest_point(line,pt);

// Generalized case (would need iterative approach and certain requirements on the function):
// returns array of points pt on line, where fn(pt, arg) has local minimum
line_closest_points_fn(line, fn, arg);
```

Again, this idea originates from the point where I didn't realize we have analytic solution for cubic beziers, so it might be just too ugly if most of real-world use cases have dedicated analytic-solution functions...

adrianVmariano, 2021-02-02:
Seems like what you're actually asking for is a 1d function minimizer.  I have pondered writing a 1d function solver and possibly also a minimizer, but I usually try to restrain myself and not write the thing until I have an actual application.  (I actually wrote the polynomial solve without an application, but we've found uses for it.  I have an eigenvalue decomposition written and no use for it.)   The case of heightfield is, I think, the only place where function literals are used right now.  

The problem with replacing functions like line_line_intersection is that it means specifying lines using equations, which is generally not natural, and it replaces a closed form direct solution with an iterative solution.  

I tried the 4th order continuous curvature solution and the extra degrees of freedom are no trouble.  They hardly change the answer at all.  

[image]

Not sure that looks much different.   Does it look better? 

So I think to close this issue we need to put the above bezier_line_intersection into the lib.  Two questions:  

1. Should it return the bezier parameter (as it does now) or the actual intersection points?
2. Should it return invalid points (outside [0,1]) like it does now, or automatically limit them.  Or make this an option?

RonaldoCMP, 2021-02-02:
> So I think to close this issue we need to put the above bezier_line_intersection into the lib. Two questions:
>
> 1. Should it return the bezier parameter (as it does now) or the actual intersection [...]

I tried to be concise in the definition of `bezier_line_intersection()` in order to keep it simple. To include it in the library it would be reasonable to introduce two independent optional arguments one for each of the two questions you posed.

A possible extension would be a function that receives a Bezier path as many other functions do.

martincizek, 2021-02-02:
> Seems like what you're actually asking for is a 1d function minimizer. I have pondered writing a 1d function solver and possibly also a minimizer, but I usually try to restrain myself and not write the thing until I have an actual application.

Yes, that would be nice. So as a user, I'm all for it. But I admit it can be controversial - let me share a piece of anecdotic evidence. I came across the need to construct a circle arc - with given radius, given point on circle and given tangent line. The center point has to be calculated. Functions like `circle_2tangents()` didn't do a trick, but it should be easy with calculus, ending up in some quadratic equation, right ... after covering two A4s with equations, I decided to use bisection method and brute force, because CPU power is cheap. So the controversy might be using solvers for tasks that do not really need it. But yes, I'd rather use a well-advised solver than my ad-hoc one.

> The problem with replacing functions like line_line_intersection is that it means specifying lines using equations, which is generally not natural, and it replaces a closed form direct solution with an iterative solution.

Understand.
> 
> I tried the 4th order continuous curvature solution and the extra degrees of freedom are no trouble. They hardly change the answer at all.
> Not sure that looks much different. Does it look better?

Wow, thanks! It seems to be slightly better. Would you mind to share the code?

> So I think to close this issue we need to put the above bezier_line_intersection into the lib. Two questions:

Nice to hear that. :)

> 1. Should it return the bezier parameter (as it does now) or the actual intersection points?

My two cents: I'd vote for keeping the bezier parameter, as it is a) more generic, b) convertible to point with other functions and c) consistent with `bezier_segment_closest_point()` and `bezier_path_closest_point()`

> 2. Should it return invalid points (outside [0,1]) like it does now, or automatically limit them.  Or make this an option?

My two cents: I'd drop the results outside the interval, as these points probably do not have any real-life meaning. It would be also sort-of consistent with e.g. `line_segment_intersection()`, which also does not return points not on the segment. :)

revarbat, 2021-02-02:
This sounds like a good thing to have.  I'll leave specific implementation to either @adrianVmariano or @RonaldoCMP.  I'd drop results outside 0<=u<=1.  Unless you think there's a good reason to return those.

revarbat, 2021-02-02:
Finding the circle center from a point on the circle, a tangent line not necessarily through that point, and a radius?
```openscad

adrianVmariano, 2021-02-02:
include<BOSL2-fork/std.scad>
    include<BOSL2-fork/beziers.scad>

    p1 = [-120,0];
    p2 = [-60, 20];
    p3 = [0,30];
    slope = .5;
    k=0.5;

    isect12 = line_intersection([p1,p1+[1,0]], [p2, p2+[1,slope]]);
    isect23 = line_intersection([p3,p3+[1,0]], [p2, p2+[1,slope]]);

There's the continuous curvature code.

I don't understand the circle problem.  If you have a line and a point and you want tangent to the line and through the point with given radius then I think there are 4 solutions and they should be pretty easy to find with geometry.  The problem with solvers is when there are multiple solutions, controlling the solutions you get.  Or if you need all of them...how do you find them all?

martincizek, 2021-02-02:
@adrianVmariano Thanks, will check it out.

@revarbat @adrianVmariano I know it's not difficult in principle - the problem can be converted to finding line-to-circle intersections[*]. My point was that introducing the generic-solver weapon might make users use it even for such tasks, where analytic solution is accessible. As a user, I don't mind. As a hypothetic author of such a smart library, I might prefer pushing users rather to ask for the feature instead of letting them hack line-to-circle intersections with an iterative solver... Which is what I did, because it was easier at the moment.

[*] Having said it is offtopic for this ticket, I really appreciate your interest! So if there is an easy way to find line-to-circle crossings with the current functions, let me know. If there wasn't, I might try to draft some contribution to pay your interest back. :)

martincizek, 2021-02-02:
> ```
> include<BOSL2-fork/std.scad>
> include<BOSL2-fork/beziers.scad>
> 
> p1 = [-120,0];
> p2 = [-60, 20];
> p3 = [0,30];
> slope = .5;
> k=0.5;
> 
> isect12 = line_intersection([p1,p1+[1,0]], [p2, p2+[1,slope]]);
> isect23 = line_intersection([p3,p3+[1,0]], [p2, p2+[1,slope]]);
> 
> color("red")move_copies([p1,p2,p3]) circle(r=1,$fn=16);
> //color("green")move_copies(bezpts) circle(r=1,$fn=16);
> 
> stroke(bezier_path(bezpts, 25, 4),width=1);
> ```
> 
> There's the continuous curvature code.

Sorry for being such a nitpicker, but how do you construct the `bezpts`? :)

adrianVmariano, 2021-02-02:
Hmmm.  I had some cut and paste issues.  I added the bezpts line with an edit to the above.

adrianVmariano, 2021-02-02:
include<BOSL2/std.scad>

    $fn=32;

    line = [rands(-10,0,2),rands(10,15,2)];
    pt = rands(0,10,2);
    r = 2;

    direction = sign(point_left_of_line2d(pt,line));

    stroke(line,width=.1,endcap2="arrow");

    cline = move(direction *line_normal(line)*r,p=line);
    d=distance_from_line(cline,pt);

    if (d>r){
      echo("no solution");
      color("red")stroke(move(pt,p=circle(r=r)),width=.05,closed=true);
      color("red")stroke(cline,width=.05,endcap2="arrow");
    }
    else
    {
      cp = line_closest_point(cline,pt)+unit(cline[1]-cline[0])*sqrt(r*r-d*d);
      move(cp)stroke(circle(r=r),width=.05,closed=true);
      cp2 = line_closest_point(cline,pt)-unit(cline[1]-cline[0])*sqrt(r*r-d*d);
      move(cp2)stroke(circle(r=r),width=.05,closed=true);
    }

    color("green")move(pt)circle(r=.1);

No need to solve any equations to solve your circle problem.

martincizek, 2021-02-02:
@adrianVmariano I owe you feedback to the curve problem - after playing with `slope` and `k` parameters, the curves really look similar, see below (yellow has continuous curvature, green is only tangent-continuous). 
[image]

And if I manually tweak the midpoint's tangent in the cubic bezier path, the paths can become even closer to each other.

For the track transition, my guess is that:
* continuous curvature and quartic beziers would feel better for vehicles that are small compared to track transition length
* cubic beziers path might work better for vehicles that are larger than the track transition length. I imagine it as if the vehicle length itself was sort-of adding one more bezier degree (I know it;s a weak idea)
* and as you correctly suggested, my case is not only about track transition :)

Thank you very much for your help! Not only I have a good-enough solution, now I'm also confident that it's really good enough and why. :)

martincizek, 2021-02-02:
> No need to solve any equations to solve your circle problem.

What to say. It's brilliant.

adrianVmariano, 2021-02-02:
I couldn't figure out how you dragged calculus into the problem...  Is it worth having a circle_line_intersection?  (One then wonders about circle_segment and circle_ray.)

martincizek, 2021-02-02:
> I couldn't figure out how you dragged calculus into the problem... 

Well, it's an issue of a "limited" user. First searching for a matching function. If not successful, then searching for an appropriate building block. For each candidate, I think if I can build the solution using the building block candidate + the things I already know. And the combination of `line_closest_point()` with `distance_from_line()` just didn't hit me at that point of library knowledge. So I temporarily jumped into solving the *line = circle* equation. Now it looks pretty straightforward. The second possible explanation is that I am just too ignorant :)

> Is it worth having a circle_line_intersection? (One then wonders about circle_segment and circle_ray.)

Depends on the users, I'm just a single sample. And with your hint, I fell into the group of those who know the simple solution.

Maybe some survey would help? I totally understand that functions for the cartesian product of all primitives would be unmanageable...

adrianVmariano, 2021-02-02:
I added line-circle and line-bezier intersection.  (PR 411)

Unless anybody has anything to add I'll close out this issue.   One observation is that the bezier_line_intersection does not work for rays or segments, but to make it do so it needs to actually calculate the points, which seems inefficient.  

I added rays and segments for the circle case.  And I kept the bezier code returning the parameter, but restricted the return to [0,1].

martincizek, 2021-02-02:
> Unless anybody has anything to add I'll close out this issue.

Nothing but applause :)

> One observation is that the bezier_line_intersection does not work for rays or segments, but to make it do so it needs to actually calculate the points, which seems inefficient.

Totally reasonable.

> I added rays and segments for the circle case.

Perfect :)

> And I kept the bezier code returning the parameter, but restricted the return to [0,1].

For the record, one more specific reason why the parameter appears to be better choice - it allows to easily get tangents and derivatives for those who need it.

I'd add one more use case for the bezier-line intersection feature, just to better justify your effort. It gives a tool for restricting the area where bezier-defined features become too thin for printing. It should help me to get rid of a few "Bulgarian constants" in my models. :)

It was a pleasure to have this discussion with you - really enlightening.

closed 2021-02-03 by adrianVmariano