BelfrySCAD/BOSL2 #362: distinguish positional parameters from named parameters

adrianVmariano, 2021-01-03:
The docs need to distinguish which parameters can be used as positional parameters and which ones must be named parameters.  

I think the best way to do this is to have the arguments block split into two blocks, a "Positional Arguments" block and a "Named Arguments" block.   We also talked about listing positional args in italics in the usage text, which I think is a good idea.  If hover text is possible that's also good.

revarbat, 2021-01-15:
Putting a `---` arguments line will split arguments between positional (before) and named (after) argument tables.
We trued putting named args in usage lines in \*italics\* but that was too subtle.  Currently trying out marking named args lie `foo=`.

adrianVmariano, 2021-11-18:
We need to nail down the style for how this is going to be done.  In about half the code named args appears with a `=` in the usage message and in the other half they do not.  Is the `arg=` form good, and what we should be using throughout?   I almost wonder if we need a kind of checklist for things to go through the code and check for, like this thing, and various other kinds of checks, whether singleton regions are handled as paths when appropriate, and so on.

closed 2022-04-12 by adrianVmariano