BelfrySCAD/BOSL2 #381: round_corners() should clamp excessive cut-offs

carstenfuchs, 2021-01-13:
**Is your feature request related to a problem? Please describe.**
round_corners() fails when cut-offs are chosen to big.
Total failure is very frustrating, especially as it seems unnecessary and is very difficult to resolve if e.g. the input path is computed.

Code to reproduce:

```
include <BOSL2/std.scad>
include <BOSL2/rounding.scad>

polygon(
    round_corners(
        [
            [0, 0],
            [10, 0],
            [20, 1],
            [20, -10],
        ],
        cut=[8, 0, 0, 0],  // works with 7, fails with 8
        $fn=20
    )
);
```

**Describe the solution you'd like**
The cut-off should automatically be clamped to the maximum value that the path permits.
Neighboring cut-offs should be weighted accordingly for a balanced resolution.

**Describe alternatives you've considered**
With paths that are less trivial than the above example, only trial-and-error is possible to find the maximum possible cut-off values.
If the path is computed, it is almost impossible to come up with cut-values that reliably work.

**Additional context**
Also see #380.

adrianVmariano, 2021-01-14:
I don't know a reasonable way to figure out what to clamp them to that works in general.  For your example with just a single roundover, of course it is easy to identify a maximum rounding.  The problem is when you have different values at every corner, how much do you change each corner to produce a valid set of parameters?  There's no unique solution to this question.   What do you mean by "neighboring cutoffs should be weighted accordingly".  According to what?   This probably turns into a system of N equations for N points.   I also don't really like the idea that your input is changed on you, because then you might say, "I want less rounding" and you decrease the number and nothing happens.   I suppose if I had an algorithm for how to do it I could allow "inf" for the maximal rounding.  

The amount of trial and error you need to use is limited by the verbose option.   For your simple case with just one roundover it will tell you that 8*.989 will work.  For more complicated cases you might have to iterate a couple times depending on what you want.  I agree that this strategy may be a problem for computed data.

carstenfuchs, 2021-01-14:
Please consider this example:
```
include <BOSL2/std.scad>
include <BOSL2/rounding.scad>

polygon(
    round_corners(
        [
            [0, 0],
            [0, 40],
            [20, 40],
            [20, 0],
        ],
        radius=[0, 12, 8, 0]
    )
);
```
The radii 12 and 8 here are at their maximum, because the length of the common edge is 20. Now, if the 8 was changed to 10, one option was to clamp the 10 back to 8, because of the edge of length 20, the first 12 units were already used by the previous point in the path.
Another option was to scale both numbers down so that their sum is 20 again. That is, use the new values 12*(20/22) + 10*(20/22). (This is what I meant with weighting.)

I understand that this can cause implicit and possibly unintended behavior such as when a single number changes, nothing seems to happen, or when a single number changes, the roundings of the neighboring points change.
However, the use cases that I have experienced so far (I'm still new to OpenSCAD) were always one of these:
  - I was hand-crafting a polygon shape, refreshing the preview after every change to a radius and besides the visual in the preview kept a good mental model of the shape. If I exceeded a radius it would have been my preference if a warning was in the console rather than the polygon vanishing from the preview – and I'd probably not even care about it, because it's more important to me that things are round than the exact amount of the roundness.
  - I was writing some code that computed the polygon shape. Almost impossible to get the radii right. In this case I'd more than happy if errors and warnings did not occur and radii were just adjusted to make them work. This too assumes that I usually don't care about the exact details of the rounding. I still closely inspect the radii, but if the algorithm fixes details for me it's much better than stopping with an error that I cannot fix.

This is possibly motivated a bit because my first encounter with rounding was https://kurthutten.com/blog/round-anything-api/, which I briefly used before changing to BOSL. At that page is also an explanation on how the author dealt with radii conflicts.

In summary, maybe the key motivation here is that rounding should be forgiving if I don't care about the details and still give me the option to control it (e.g. via console output) when I do.

adrianVmariano, 2021-01-14:
Yes, round_corners was inspired by the round-anything lib, but I did not like the hack for dealing with conflicting radii.  

The problem with the idea of radii adjusted to "make them right" is that what's right depends on user intentions.    If I specify the same rounding at every corner and only one is in violation what do I want?  Maybe I want all of them decreased so that the rounding is the same everywhere.   If the bad corner is very bad then you mysteriously get what looks like no rounding at all.  I'd say that's pretty confusing behavior.  So maybe I just want to tweak the rounding at the "bad" corner.   

The idea of adjusting parameters in a scaled way may or may not be what the user wants.  But also it's nontrivial to actually do it.   Your example above hides the complexity because you are using radius on right angle corners.  In the general case, you don't know how to adjust the parameters to get valid parameters.   The scalefactors get you closer, but don't actually work robustly in my experience.   Maybe one could use an iteration, take scalefactors and set everything above 1 to 1 and then apply the new factors and repeat until (hopefully) reaching a fixed point.  This will fail on the collinear case, I think.  And it's hard to understand what exactly it will do.  The key point is that the round_corner behavior needs to be predictable, so the user knows what it's going to do.  Its behavior should be well-defined.  Mysterious adjustment of the user's input breaks this.   Note that you say "the algorithm fixes details" but if it changes rounding from 10 to 1 is that really a "detail"?  

It seems like part of the problem is that the real intention of this approach for rounding is not to "maximally" round something, but to do what the name says and round off the corners.   Maybe your problem is better solved with smooth_path() instead.   That approach does maximally round (though it operates on edges instead of vertices).  Another possible approach would be to use turtle() and use the "arcleft" and "arcright" commands to draw the specific roundings you want.   This will naturally produce maximal roundings.   Or perhaps path_join().  

If you want to stick with round_corners and you want lots of rounding, don't care about the details, one possible solution to the problem you are dealing with is to use the "joint" parameter instead of "cut" or "radius".  It is easy to determine validity of the rounding with this parameter.  (At the moment it doesn't work with circular rounding, but that will be fixed soon.)   If you use "joint" you can easily compute a set of valid maximal rounding parameters yourself from your polygon by using its side lengths, for whatever style of maximal you favor.   Have you made use of the `verbose=true` argument to get the scale factors?

adrianVmariano, 2021-01-16:
Not going to implement this.  I could possibly see a function for calculating some kind of rounding factors for a given path external to round_corners, so you could write `round_corners(path, joint=cool_rounding(path))` but I don't know how to write the cool_rounding function.

closed 2021-01-16 by adrianVmariano