BelfrySCAD/BOSL2 #417: Shorthands for XYZ axes swapping

martincizek, 2021-02-14:
**Underlying problem:**
2D to 3D extension operations along a cartesian axis usually take 2D input in the XY plane with extrusion/extension to the Z axis. E.g. `linear_extrude()` and `skin()`. But real-world use cases often need 3D extrusion along X or Y axis. In these cases, I begin the same way, i.e. with preparing the 2D shape in XY. The pain starts when the Z-extruded shape is about to be transformed to other axes:
* For X-extrusion, I'd like to see the former X+Y+ to become Y+Z+.
* For Y-extrusion, I'd like to see my former X+Y+ to become X+Z+.
AFAIK, neither transformation is solvable by a simple 1-axis rotation. The former can be solved by multiaxis rotation, the latter needs also chaining with either translation or axis mirroring.

**Describe the solution you'd like**
Currently I'm using these two transformations to address the problem without brain twisting. I think something like that might be useful in the library.

``` scad
module transform_xyz_yzx() {
  multmatrix([[0, 0, 1, 0], [1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 0, 1]]) {
    children();
  }
}

module transform_xyz_xzy() {
  multmatrix([[1, 0, 0, 0], [0, 0, 1, 0], [0, 1, 0, 0], [0, 0, 0, 1]]) {
    children();
  }
}
```

**Describe alternatives you've considered**
* For simple linear extrusion, I've tried `extrude_from_to()`, but it does not address the problem, as it still needs extra operations to achieve axes swap.
* In particular, `transform_xyz_yzx()` can be achieved with combination of `orient` and `spin` parameters, but neither this seems straightforward  to me.
* I admit that this is redundant from the geometry point of view. It's more about human-friendliness of entering the input data or shapes.

revarbat, 2021-02-14:
You can already do this with `rot(from=UP, to=RIGHT) ...` or `rot(from=UP, to=RIGHT, a=spin)`

revarbat, 2021-02-14:
For shorthand, what about:
- `orient(vec) ...`  which is equivalent to `rot(from=UP,to=vec) ...`
- `orientx() ...` which is equivalent to `rot(from=UP,to=RIGHT) ...`
- `orienty() ...` which is equivalent to `rot(from=UP,to=BACK) ...`

revarbat, 2021-02-14:
> - For X-extrusion, I'd like the former X+Y+ to become Y+Z+.
> - For Y-extrusion, I'd like the former X+Y+ to become X+Z+.
>
>AFAIK, neither transformation is solvable by a simple 1-axis rotation. The former can be solved by multiaxis rotation,
>the latter needs also chaining with either translation or axis mirroring.

`yrot(-90) ...` or `rotate([0,-90,0]) ...` for the first.
`xrot(90) ...` or `rotate([90,0,0]) ...` for the second.

revarbat, 2021-02-14:
Hmm, based on the matrices you give, are you specifically trying to transform so the sign of all axes remains the same after rotation? That's actually far less intuitive to me than just tilting over the shape.

However, the can be implemented thus:
```openscadmodule transform_xyz_yzx() rot(v=[1,1,1], a=120)`
- `transform_xyz_xzy()` can be done with `yflip() xrot(90)`.

revarbat, 2021-02-14:
Actually, even simpler:
```openscad
module transform_xyz_yzx() rot(120,v=[1,1,1]) children();
module transform_xyz_xzy() mirror([0,-1,1]) children();
```

revarbat, 2021-02-14:
At this point, the proposed shortcut names are longer than the equivalent implementation.  Will need shorter names.
`xyz_to_yzx()` and `xyz_to_xzy()`?  Eh. This feels like it implies the existence of modules for all the other `xyz` permutations.
`rot_axes()` and `swap_yz()`? maybe clearer, but they don't sound related.

revarbat, 2021-02-14:
Actually, the `swap_yz()` could be `yzflip()` and be added to the mirroring transforms, along with `xyflip()` and `xzflip()`.  That seems elegant.

revarbat, 2021-02-15:
That in turn implies `xyzrot()`, like `xrot()`, `yrot()` etc.

martincizek, 2021-02-15:
> Hmm, based on the matrices you give, are you specifically trying to transform so the signs of all axes remains the same after rotation?

Correct, except...

> That's actually far less intuitive to me than just tilting over the shape.

...except for the fact that the "rotation" is not the actual intent. The intent is to "enter my inputs in the common X+Y+Z+ sense". The XY shape extruded to Z is just an intermediate step and implementation detail. I mean - my user view is e.g. "I entered YZ 2D values and extruded them to X", not that "I created a Z-extruded shape and rotated it". Is it that uncommon to think about the model like this?

> Actually, the swap_yz() could be yzflip() and be added to the mirroring transforms, along with xyflip() and xzflip(). That seems elegant.

I knew you guys always come with something smart. Totally agree :)

revarbat, 2021-02-15:
Implemented in commit 61fa62e68a7e766e9000f10d7d9fe72ea5f51cec

You can now use `xyzrot(120)` and `yzflip()` for your two requested transforms.

adrianVmariano, 2021-09-11:
I want to remove xyzrot() and also xyrot(), xzrot() and yzrot() because I think they aren't really very intuitive/useful shortcuts and I'm trying to decrease clutter in the library.   Instead of xyzrot(120) the desired transform can be more comprehensibly obtained via frame_map(x=[0,1,0], y=[0,0,1]), or frame_map(x=BACK, y=UP), which explicitly maps the axes in the desired manner instead of via a rotation that is somewhat harder to grasp.     martincizek can you comment?

martincizek, 2021-09-11:
@adrianVmariano Fine with me. Actually I still use my shorthand modules like `xyz_to_yzx()`, as these are more readable for me. According to the docs for `frame_map()`, it seems to address the same use cases I've mentioned above. I like the design!
And thanks for asking btw.

closed 2021-09-11 by adrianVmariano