BelfrySCAD/BOSL2 #431: [BUG] documentation: missing information for metric nut()

3Dietrich, 2021-02-21:
**Describe the bug**
 I re-downloaded BOSL2 right now and I researched for a while to figure out how to get a metric nut out (and not long bug reports). So I consider this as a documentation bug:
There is a problem in describing [metric nuts](https://github.com/revarbat/BOSL2/wiki/screws.scad#module-nut) :

1. the description of Module: nut() misses the "diameter' in the argument list! Without setting a diameter you get meter long bug reports (not helpful for me)
2. there is not any example for metric nuts [on the whole page](https://github.com/revarbat/BOSL2/wiki/screws.scad)!
3. if I found the correct inputs, the M6 nut and bolt doesn't fit at all..

**Code To Reproduce Bug**
getting a metric screw is easy
```
include <BOSL2/std.scad>
include <BOSL2/screws.scad>
screw("M6,22");
```
but not a nut:
```
include <BOSL2/std.scad>
include <BOSL2/screws.scad>
screw("M6");
// results in long error
```
There is some infos missing (see Expected behavior below)

however, if you put both (correctly?), no one can tell me that the nut and the bolt fit together:
```
include <BOSL2/std.scad>
include <BOSL2/screws.scad>
difference() {
	union() {
		screw("M6,12");
		nut("M6", thickness=12, diameter=10);
	}
	cuboid(50, anchor=BACK);
}
```
Viewed from above, you can see that there is a gap of approx. 0.2mm between the outer bolt and the inner nut! What did I enter wrong?

**Expected behavior**
1. you need for a metric nut at least one example, like this (minimal entries):
```
include <BOSL2/std.scad>
include <BOSL2/screws.scad>
nut("M6", thickness=6, diameter=10);
```
2. shouldn't the **Usage:** be written like this:
nut(name,[thread],[oversize],[spec],diameter,thickness,[tolerance],[details])
I.e. '[]' means "optional" but without it's "mandatory"?

3. ```diameter``` should also be metioned in "Arguments" (even if it has been described somewhere before)..
4. last but not least: if the thread thicknesses are the same, the bolt and nut should fit correctly into each other

adrianVmariano, 2021-02-21:
Note, the 0.2mm error you find is a result of the $slop parameter, which defaults that value.  Set it smaller and things should work better, or set it to zero if you want the exact correct dimensions, where nut fit is determined solely by the ISO tolerance rules with no extra gap.  This $slop parameter is used throughout the library and for my printer the default value is much too large.  

The screw library is still a little rough, and think in particular nuts have not been well considered.   (Effort all went so far to getting screws right.)   The diameter and thickness for nuts should actually be optional, with defaults coming from standard nut dimensions.  But we don't have tables of standard nut dimensions in there yet.   Also "diameter" may be the wrong term for a measurement that isn't really a diameter.  Anyways, for now I've explicitly documented $slop on both screw() and nut(), added some nut() examples (there were none) including one that shows a metric nut and a UTS nut.  

If you feel motivated to look up and provide a table of standard nut dimensions for the metric case we can add it.  :)   But for now once my PR is integrated at least the docs should match behavior.

3Dietrich, 2021-02-21:
ok. thank you!
My problem is always to find a screw library for OpenSCAD. And BOSL2 would be the "number 1" for me for this. I was happy that you went on and now bolts/nuts were added, excellent! But it must also work sometime, after 4 hours of reading/studying, trying..

Maybe I'm hanging out a bit far, but: I never heard about "$slop". Until right now from you directly. But this seems really important if this makes the thread unusable. Is there an Page you describe this? Definitely not on this page [metric_screws.scad](https://github.com/revarbat/BOSL2/wiki/metric_screws.scad) (what I would expect).

adrianVmariano, 2021-02-21:
I don't know if/where Revar documented $slop.  I'm not sure that $slop is the "right answer", because I don't know that every part needs it the same.   And it's really easy to forget it's there and get a surprise.   And the default is much too big for my printer.  Maybe the default should be zero.  This concept is due to Revar.

3Dietrich, 2021-02-22:
yes, $slop is the right answer for this "bug". 

[image]

 to "fits 100%" to the ISO (added ```$slop = 0;```)
[image]

You added already everything - **great!** 
One little thing still:
in **Module: nut() -- Usage:** you wrote:

> nut([name],diameter, thickness,[thread],[oversize],[spec],[tolerance],[details],[$slop])

but it should be like:

> nut(name,diameter, thickness,[thread],[oversize],[spec],[tolerance],[details],[$slop])

(as you showed in the 'Example 1' it needs the name="M8")

adrianVmariano, 2021-02-22:
Actually name is supposed to be optional because you can give spec instead.

3Dietrich, 2021-02-22:
i see

revarbat, 2021-02-22:
This may be a case for using two Usage: lines.  One for two main ways to specify a nut.
Also, I can see the justification for `$slop=0`, especially with resin printers.  I only added it because it was the only way I saw to get my joiners to mate correctly on my sloppy FDM/FFF printers.

3Dietrich, 2021-02-22:
yea, I definitely see the benefit of $slop! It was just not mentioned at all.
I think @adrianVmariano has solved everything and I close the bug now.
thank you!

closed 2021-02-22 by 3Dietrich