BelfrySCAD/BOSL2 #448: A better 2D hull function

RonaldoCMP, 2021-02-28:
I have found a 2D hull function faster than the one implemented by `hull2d_path()`.  I based may implementation on a C code referenced in a code comment. It performs a previous lexicographical sort of the input points and operates in two steps: initially it finds the lower points of the convex hull and then proceeds to the upper part of it. The implementation runtime seems to be linear with the number of input points for less than 300 points but I would expect that it has the optimal complexity.

I attached a file with the code implementation of the process including a random test of it and a runtime comparison with `hull2d_path()` (change the extension to .scad to run it). For 200 points, it is 5 times faster than the current implementation in BOSL2.

[BOSL2_hull.txt](https://github.com/revarbat/BOSL2/files/6056979/BOSL2_hull.txt)

revarbat, 2021-02-28:
Hmm.  The file you link to is identical to the current BOSL2 hull.scad file.  Are you sure you linked to the correct file?

RonaldoCMP, 2021-03-01:
Diacho! I have attached the wrong file. Sorry. Here is the right one.

[hull_points_2_BOSL2.txt](https://github.com/revarbat/BOSL2/files/6058887/hull_points_2_BOSL2.txt)

adrianVmariano, 2021-03-04:
Looks like the new one returns the actual points instead of an index list.  Can you change it to return an index list?  I think that's better.  (Also easier to check that results match output from old code.) 

I tested with 20k points and the new code took 3s compared to 59s for the old code.   New code was still showing linear runtime.  It handles 200k points in 39s.

RonaldoCMP, 2021-03-04:
Here is the revised code, now returning the indices of the points in the convex hull.
I have included a check to confirm the new code returns the same polygon returned by `hull2d_path()`.

[hull_points_2_BOSL2.txt](https://github.com/revarbat/BOSL2/files/6085640/hull_points_2_BOSL2.txt)

adrianVmariano, 2021-03-04:
It looks good to me, and clearly faster.  I integrated it and found it failed a test for collinear inputs.  The old version handled this case differently, and also the cases of <3 points.  I adjusted it to match the old version and included it in my latest PR.

RonaldoCMP, 2021-03-06:
With a little change in the function `_is_CW()` , the function `hull2d_path()` is itself able to deal with the collinear cases:

`function _is_cw(a,b,c) = cross(a-c,b-c)<-EPSILON;`

Besides, that will eliminate the points on the hull border whose adjacent edges are (virtually) collinear (like the old function does).

So, the preamble of `hull2d_path()` could be just:
```
function hull2d_path(points) =
    assert(is_path(points,2),"Invalid input to hull2d_path")
    len(points) < 2 ? [] :
    let( n  = len(points), 
         ip = sortidx(points) )
    // lower hull points
    ...
```

RonaldoCMP, 2021-03-06:
To be scale independent, a better definition would be:

`function _is_cw(a,b,c) = cross(a-c,b-c)<-EPSILON*norm(a-c)*norm(b-c);`

BTW, `_hull_collinear()` fails with 

`points = [[-10,0], [-10,0], [10,0] ]`

adrianVmariano, 2021-03-06:
The proposed change fails on line 9 of tests/test_hull.scad.

RonaldoCMP, 2021-03-06:
Yes, but the hull is the same:

`hull(test_collinear_2d) == [1,7]`

The assert should be rewritten.

adrianVmariano, 2021-03-06:
That's not what I'm getting.  

     ECHO: "EXPECT: [7,1]"    
     ECHO: "GOT   : [1,2,3,4,5,0,6,8,7,8,6,0,5,4,3,2]"

adrianVmariano, 2021-03-06:
function _hull_collinear(points) =
    let(
        a = points[0],
        i = max_index([for(pt=points) norm(pt-a)]),
        n = points[i] - a
    )
    norm(n)==0 ? [0]
    :
    let(
        points1d = [ for(p = points) (p-a)*n ],
        min_i = min_index(points1d),
        max_i = max_index(points1d)
    ) [min_i, max_i];

I think this fixes the problem with _hull_collinear.

RonaldoCMP, 2021-03-06:
Indeed but it would be simpler:
```
function _hull_collinear(points) = 
    let( ps = sortidx(points) ) [ps[0], ps[len(ps)-1]];
```

adrianVmariano, 2021-03-06:
I am not so sure.  First of all sort is O(n log n) and finding the max is O(n).  But also, sort seems like it can choose a poor option, such as if the inputs are [0,0,0], [0,0,10],[0,0,100],[1e-9,0,0].  I show them in sorted order.  The third point is very close the the first, but sorts last because its first index is epsilon larger.

RonaldoCMP, 2021-03-06:
You are right. Clever counterexample.