BelfrySCAD/BOSL2 #456: cheat sheet is not cheat sheet anymore

3Dietrich, 2021-03-02:
**Is your feature request related to a problem? Please describe.**
I remember a pretty useful cheat sheet some months ago. I have favorited the link, but.. right now it's comeletely different and doesn't help me. It's something like a very long complex data sheet. When I search something like "elipse" I used to be able to look through the list and then find "cycle" or "oval". Right now I have no chance, I need to use the text search. And don't find the solution.

**Describe the solution you'd like**
It should be an gate in short words linked into the deeper topics. Like it was before. A simple overview linking into the details.

**Describe alternatives you've considered**
The cheat sheet as it is now certainly helps for some cases.., it can be still there but with a different name (and a link in the new/old cheet sheet :)

revarbat, 2021-03-02:
The current Cheat Sheet is an attempt to make it closer to the OpenSCAD Cheat Sheet, which gives function names, typical uses, and arguments.  I'm also currently in the process of adding Topics headers to everything, which generates a Topics.md file sorted by topic name, which sounds more like what you are looking for, I think.

revarbat, 2021-03-02:
The current, incomplete, Topics index is at:
https://github.com/revarbat/BOSL2/wiki/Topics

3Dietrich, 2021-03-02:
but.. don't you see the difference?
OpenSCAD's cheat sheet is an perfect example for an good cheat sheet! It's short and clear.
Now go and watch the [BOSL2 cheat sheet](https://github.com/revarbat/BOSL2/wiki/CheatSheet)  - it's sooo loooonng. every(!) possible entry is repeated and repeated (with changing some details..) That's not an cheat sheet, that's a maimum premium book edition..

revarbat, 2021-03-02:
Hmm.  If you just want the function/module/constant names linked, without arguments, exactly once, isn't that the same as either the TOC.md (table of contents), or the Index.md (alphabetical index)?

For the record, these are the existing indices:

https://github.com/revarbat/BOSL2/wiki/TOC
https://github.com/revarbat/BOSL2/wiki/Index
https://github.com/revarbat/BOSL2/wiki/Topics
https://github.com/revarbat/BOSL2/wiki/CheatSheet

revarbat, 2021-03-02:
Part of the problem is that BOSL2 is a rather large library, which makes for long indices.
The idea of the TOC index is to list things by file and section.
The idea of the alphabetical Index is to let you look up any function/module/constant by name.
The idea of the Topics index is to let you look for a topic of interest and find what calls are available for that.
The idea of the CheatSheet is to give you a page to show typical uses of everything, with typical args.  For a number of functions/modules there are multiple different typical usages.  Right now, the Cheat Sheet is created from the Usage: documentation lines for each function/module.

3Dietrich, 2021-03-02:
well.. maybe I remember an phantom. Wasn't there a cheat sheet with an clean short overview a few months ago? Maybe I'm wrong or the lib is grown extensily.

3Dietrich, 2021-03-02:
anyway, it's a great lib with great manual and obviously many approaches to find the things you want. Thank you for that!

revarbat, 2021-03-02:
The only previous version of the Cheat Sheet just listed the function/module names for each file, with no arguments.  Kind of like the current TOC.md.  Except perhaps more compactly.  I can work on that.

revarbat, 2021-03-03:
I've made tweaks to TOC.md and CheatSheet.md to make them more compact.  Hopefully this makes things more reasonable.

3Dietrich, 2021-03-03:
yes!!
thank you so much!!

3Dietrich, 2021-03-03:
it's almost a bit of a pity to close the bug, I would still like to celebrate it :)

closed 2021-03-03 by revarbat