BelfrySCAD/BOSL2 #457: How can I create a rounded box where the inner walls are heightfields?

zumpchke, 2021-03-03:
How can I get my heightfield surface to follow the inner surface exactly?

adrianVmariano, 2021-03-04:
I don't understand the question.   You want the outside of the box to be a cylinder but the inside to follow a 2d height field wrapped onto the cylinder?

zumpchke, 2021-03-09:
@adrianVmariano 
Basically I want a box with a bumpy sinusoidal texture (think acoustic surface).

[image]

Here, I have used the example of connecting bezier surfaces in the docs to give you an idea of what I'm talking about. But it's not very robust since the bumpy surface abruptly stops and does not follow the rounding.. any ideas if I can accomplish this with heightfield?

revarbat, 2021-03-09:
Heightfield won’t do this.  You’re actually following the best approach here, I think.  Adding the sinusoidal bumps to the edge patches should be doable, I think, if you subdivide the patch into a bunch of smaller ones.  The corners may be more difficult, due to the degenerate edge, but I think it should be doable.

- Revar

adrianVmariano, 2021-03-09:
You could use heightfield if you used it to design one face of the cube, including half the roundover, so that you could then assemble six of them together.  But that seems unlikely to be the easiest way to do it.  I think if you have some sine wave shape you want that using the vnf commands to just directly assemble your shape is probably the best route.  You will be able to calculate vertex positions and go around the curves as required, and you can do the whole thing in a continuous manner which should make it possible to handle the boundary conditions....though I do wonder if you'll run into some problems somewhere with that.

closed 2021-03-11 by zumpchke