BelfrySCAD/BOSL2 #472: [BUG] worm_gear: invalid polyhedron (CGAL error at render when intersecting)

wosym, 2021-03-14:
**Describe the bug**
When drawing a worm gear and then intersecting it with another object (e.g. a hole for an axle) the preview works, but the render does not. It throws the following error:
```ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /usr/include/CGAL/Nef_S2/SM_const_decorator.h Line: 330```

**Code To Reproduce Bug**
```
worm_mod = 1;
worm_d = 10;
worm_gear_t = 20;

difference() {
    worm_gear(mod=worm_mod, teeth=worm_gear_t, worm_diam=worm_d, orient=LEFT);
    cube([10,10,10]);
}
```

**Expected behavior**
Render without errors

**Additional context**
- OpenSCAD Version: 2019.05
- BOSL master: 5faecd6

@adrianVmariano noted the following:
> In the gear test posted above the polyhedron has a degenerate face = [6,7,6]. This causes debug_faces() to fail. Should this be reported as a defect instead? (Is it a defect?)

wosym, 2021-03-15:
Thanks to the great work of @revarbat  this got fixed in #473 . Unfortunately, I noticed that the issue also exists in `worm()`. 

> ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /usr/include/CGAL/Nef_S2/SM_const_decorator.h Line: 330

revarbat, 2021-03-15:
Worm is fixed, even if I'm not completely happy about the triangulation.

closed 2021-03-15 by revarbat