BelfrySCAD/BOSL2 #482: WARNING: Ignoring unknown function 'first_defined' in file BOSL2/strings.scad,

nohkumado, 2021-03-23:
Hello!

could you post somewhere the code to that module? that error message, and the subsequent host of warnings like
     WARNING: undefined operation (undefined <= number) in file strings.scad, line 328
 grow ever more annoying the bigger my project becomes...
thanks in advance!

revarbat, 2021-03-24:
`first_defined()` is in BOSL2/common.scad

revarbat, 2021-03-24:
You should get common.scad when you do `import <BOSL2/std.scad>`

nohkumado, 2021-03-24:
Indeed! That works !
Thanks!
Heh darn :D does this return undef by design, or could this be made
null-safe?

Ciao

Bruno Böttcher
--
35 rue de la république, FR-6720 Schwindratzheim
email: ***@***.***, ***@***.***
Fon:+33 3 88 89 91, Mob:+33 6 76 55 82 68
-------------------------------------------------
Dev: Java/Perl/PHP OS:GNU/LINUX, Android

revarbat, 2021-03-24:
If you have a number last in the list, it will be the fallback value.  ie:
```openscad
x = first_defined([foo, bar, 23]);
```
If `foo` and `bar` are both `undef`, then it will fall back to returning `23` to `x`.

closed 2021-03-25 by revarbat