BelfrySCAD/BOSL2 #486: [BUG]

LenStruttmann, 2021-03-29:
It looks like I've run across some apparent issues in BOSL2/gears.scad

When using anchor=TOP" or BOTTOM, or "flattop" in bevel_gear(), the resulting translation is incorrect:

When using anchor=TOP or "flattop", the gear top is just above the axis.
When using anchor=BOTTOM, the gear bottom is just below the axis

**Code To Reproduce Bug**
```
$fn = 24;

include <BOSL2/std.scad>;
include <BOSL2/gears.scad>;
include <BOSL2/attachments.scad>;

#bevel_gear(
   pitch=5, teeth=36, face_width=20, shaft_diam=5,
   pitch_angle=45, spiral_angle=0, anchor=BOTTOM  // or TOP, or "flattop"
)
```

**Expected behavior**
For anchor=BOTTOM, I expect the bottom gear face to be aligned with the XY plane.
For anchor=TOP or "flattop", I expect the top gear face to be aligned with the XY plane.

A secondary consequence of this is that the true thickness of the gear is not available to the user.

When using anchor="apex" to get a correctly matched pair of bevels, as far as I can tell, the user MUST have access to the true thickness and/or the distance from the apex to (at least) one face of the gear to correctly place the gears in an assembly.  This suggested that one or more bevel-gear informational functions may need to be added.

This issue seems to be caused by the fact that the "thickness" variable is calculated differently in the bevel_gear() function...

Line 963:         thickness = abs(verts1[0][0].z - select(verts1,-1)[0].z),

...than it is in the bevel_gear() module...

Line 1028:     thickness = face_width * cos(pitch_angle);

adrianVmariano, 2021-03-29:
The quoted example fails render.

LenStruttmann, 2021-03-29:
Sorry, the example is missing a semicolon ";" at the end.

it should be

$fn = 24;

include <BOSL2/std.scad>;
include <BOSL2/gears.scad>;
include <BOSL2/attachments.scad>;

#bevel_gear(
   pitch=5, teeth=36, face_width=20, shaft_diam=5,
   pitch_angle=45, spiral_angle=0, anchor=BOTTOM  // or TOP, or "flattop"
);

revarbat, 2021-04-19:
At this point I cannot reproduce the CGAL errors.

LenStruttmann, 2021-04-19:
Here is an even simpler example:

**include <BOSL2/std.scad>;
include <BOSL2/gears.scad>;

bevel_gear( anchor=BOTTOM );**

The resulting gear is NOT anchored at the gear bottom.

Similarly, using anchor=TOP, the resulting gear is NOT anchored at the gear top.

revarbat, 2021-04-19:
The thickness is definitely incorrect.  I am unclear where the extra thickness is coming from mathematically, but I can at the least force it to the expected size.

revarbat, 2021-04-19:
Fixed in 6687b4cafcbabcecd28e8a36c1bb409bd9294815

closed 2021-04-19 by revarbat