BelfrySCAD/BOSL2 #489: round_corners and path_chamfer_and_rounding do the same thing

adrianVmariano, 2021-04-03:
Well, they almost do the same thing.   It seems confusing to have two separate functions.  If we actually want/need them it seems like path_chamfer_and_rounding should be in rounding.scad with round_corners and an explanation of why there are two different functions.

The way in which they differ is that path_chamfer_and_rounding can do chamfers and roundings on the same path whereas round_corners can only apply one method at a time.   Other than that, round_corners is more flexible, providing different ways to specify roundings and chamfers, and supporting the "smooth" mode.  

I'm not sure why _corner_chamfer_path is so complex.   The equivalent code from round_corners (_chamfcorner) is only 4 lines of code.   Maybe it needs an internal doc text.  It looks like _corner_chamfer_path was written to do some much larger case of chamfering than it's used for, but it's only called from path_chamfer_and_rounding, so what's the point?  

Of course _corner_roundover_path also duplicates the function of _circlecorner from rounding.scad but without supporting the 3d case.

revarbat, 2021-04-05:
`rect()` and `prismoid()` rely on `path_chamfer_and_rounding()` to do rounding and/or chamfering per corner, mixing and matching chamfers and roundings in the same shape.  I originally wrote the function before remembering about `round_corners()`, to be honest, but I like the per-corner selection of roundings and chamfers.  If we can get that functionality in `round_corners()`, we can remove `path_chamfer_and_rounding()`.

revarbat, 2021-04-05:
Oh, `trapezoid()` uses `path_chamfer_and_rounding()` too.

adrianVmariano, 2021-04-05:
It's not obvious how to make round_corners handle both chamfers and roundings on the same curve in a fully flexible manner.  We could allow it to take an array for `method` but then things get potentially messy unless you require that `cut` or `joint` be used, because those are the only metrics that work for all methods.   I guess it could work to allow both `joint` and `radius` (for example) to be given and then you'd have to check that at most one is nonzero at every index and the nonzero one was allowed for the method.  It seems a little complex.  I guess you have a similar check in the other function so someone doesn't try to round and chamfer the same corner?

adrianVmariano, 2021-04-05:
Would it be flexible enough for your needs if you had to give the rounding/chamfers using the "joint" parameter?   I think this is equivalent to the "side" for chamfers.  But you'd need to calculate radius -> joint for roundings. 

You'd need to pass a list of methods like `["chamfer", "chamfer", "circle", "circle", ...]`   That seems a little bit awkward.

I think making round_corners work this way wouldn't be too messy, though there's the question about expanding `method` into a huge list of strings.  It would be messier to allow mixing `joint`, `radius` and `cut`.

revarbat, 2021-05-27:
Note: If we alter `rect()`, `trapezoid()`, and `prismoid()` to use `round_corners()`, then we will need to include rounding.scad in std.scad.

revarbat, 2021-07-11:
Or else we will need to move `round_corners()` out of rounding.scad into paths.scad.

adrianVmariano, 2021-07-11:
That's true, though it seems sort of arbitrary unless you move all the 2d functions into paths.scad: smooth_path, path_join, and offset_stroke.

closed 2022-01-20 by adrianVmariano