BelfrySCAD/BOSL2 #511: [BUG] Cannot have a guided half_joiner2 with an unguided one in the same model

pfelecan, 2021-04-17:
Cannot have a guided half_joiner2 with an unguided one in the same model as in the following snippet:

```
	 half_joiner2(h = 15, l = 10, w = 7, screwsize=3);
	 right(20) half_joiner2(h = 15, l = 10, w = 7, screwsize=3, guides = false);

```

Here is the output when rendering the model:

```
   OpenSCAD 2021.01
   https://www.openscad.org/

   Copyright (C) 2009-2021 The OpenSCAD Developers
   This program is free software; you can redistribute it and/or modify it under the terms of the GNU General Public License as published by the Free Software Foundation; either version 2 of the License, or (at your option) any later version.

   Loaded design '/pariNAma/3Dprint/JoinersTest.scad'.
   Compiling design (CSG Tree generation)...
   Compiling design (CSG Products generation)...
   Geometries in cache: 55
   Geometry cache size in bytes: 125760
   CGAL Polyhedrons in cache: 27
   CGAL cache size in bytes: 1706576
   Compiling design (CSG Products normalization)...
   Normalized tree has 2 elements!
   Compile and preview finished.
   Total rendering time: 0:00:00.549

   Parsing design (AST generation)...
   Compiling design (CSG Tree generation)...
   Rendering Polygon Mesh using CGAL...
   ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e_below != SHalfedge_handle() File: /usr/include/CGAL/Nef_3/SNC_FM_decorator.h Line: 418 
   ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e_below != SHalfedge_handle() File: /usr/include/CGAL/Nef_3/SNC_FM_decorator.h Line: 418 
   Geometries in cache: 57
   Geometry cache size in bytes: 137720
   CGAL Polyhedrons in cache: 27
   CGAL cache size in bytes: 1706576
   Total rendering time: 0:00:00.172
   WARNING: No top level geometry to render 

```

adrianVmariano, 2021-04-18:
It looks like half_joiner fails CGAL.  It doesn't even have any obvious way to fail, being just an assembly of cubes.  

Note that it's not that you used two joiners that triggered the error, but that you had a joiner and anything else.  If you combine a joiner with a cube it fails.

revarbat, 2021-04-19:
I changed the construction of `half_joiner()` and that seems to fix this.

closed 2021-04-19 by revarbat