BelfrySCAD/BOSL2 #521: vnf_vertex_array styles: when to use which ones (examples), quincunx and concave

adrianVmariano, 2021-04-25:
I'm wondering about the styles for vnf_vertex array.  Are they all useful?  And if so, should we add a concave style?  I think we need some examples showing cases where the different VNF styles work best, and how the wrong choice looks bad.  

Is there ever a situation where the quincunx style produces a good (desirable) result?   Seeing the example with gears where it made things look bad made me aware of this issue: it introduces a new point which is only actually on the polyhedron if the face being split is coplanar.   So it seems like we have a situation where if the face is coplanar this is a good choice....but it won't look better than the alternative.  If the face is nearly coplanar it should also be a decent choice, but it probably still won't look better.  And if the face is far from coplanar it's going to twist the shape in a weird way and the result will be bad.   I'm inclined to think that we should remove this style.  

A similar question arises for the convex style.  Is there some case where it produces the best result and is different than the other results?  Presumably this would be for convex shapes? If such an example exists then presumably it can be inverted to require the concave style, so then presumably the concave style should be implemented as well.

So I'm thinking that "quincunx" should go away.  And either we add "concave" or we also get rid of "convex".  

Whichever styles we keep, there needs to be some examples in the docs showing how the style differ and when the various styles look good/bad.

revarbat, 2021-04-25:
"quincunx" was added for the case where the surface flexes both ways, making both the default and "alt" styles incorrect for parts of the same surface. Consider passing `function (x,y) sin(x/3) * sin(y/3)` to `heightfield()`. You get places on the surface where it naturally wants to bend against how it gets triangulated. 

It preceded "min_edge" and "convex". I'm fairly sure "convex" doesn't make a better look for this case. Nor would a theoretical "concave", which we maybe should implement. Dunno about "min_edge".

adrianVmariano, 2021-04-25:
I'm having a hard time figuring out whether quincunx actually looks better for that case.  The problem is that it's hard to know how to make a good comparison.  I tried doing difference of a high resolution version of your height field and differencing with low res using different styles.  By this measure, quincunx performed worse than everything except convex.  

Another observation is that quincunx doubles the number of triangles.  Presumably it would be better to double the triangles with real data instead of interpolated data.  I guess the key observation is that what quincunx is trying to do is make a grid based on triangles instead of quads, by having alternately offset rows of points.  And whether it works well or not depends on how good its approximation for the interpolated points actually is.  So like perhaps something more sophisticated than linear interpolation would work better?  

Now I did identify a case where quincunx clearly works very well:  in refined sweeps the refinement is linear which means that quincunx is calculating the exact centerpoint, so you get an extra point added exactly.  (Does this mean quincunx should be the default for skin?)  

    skin([path1,path2], z=[0,5], refine=30, slices=30);
    color("red")vnf_wireframe(skin([path1,path2], z=[0,5],  slices=0, style = "quincunx") ,d=.1);

But in the height field example, this is not the case.  Though quincunx does produce symmetry.

amatulic, 2025-05-08:
I see no reason anymore to keep quincunx.

Intuitively I had always thought quincunx would improve the appearance due to artificially increasing the resolution of the mesh. As I learned when I implemented contour() using interpolation on a quincunx, it just increase the roughness of the contour line, although two passes with a 3-point box filter smoothed things out so that the appearance was better than making a contour through a 4-point quad.

I observe similar effects on 3D surfaces. The problem is that **every** quincunx center point is a saddle point. That makes every quad into a saddle, giving a rougher appearance. The vast majority of the time, quads aren't saddles, but they're part of a ridge, valley, or compound curve that's convex or concave in all directions.

A few weeks ago I experimentally added a new triangulation method to vnf_vertex_array. It split a quad into two triangles, but the choice of diagonal corners from which to create a new edge depended on the vertex normals; basically two diagonal corners would be connected by an edge if the angle between the two normals is less than the angle between the normals of the other two diagonal corners. This did result in triangulation directions based more on local geometry, but subjectively I didn't see a noticeable improvement over the other available methods, so I never submitted a PR for it.

The *reason* it didn't look like an improvement was because the direction of the new diagonal edge in each quad almost never aligns with the actual curvature of the surface. It worked well for surface ripples oriented 45° left or right, but not much else.

Another thing I considered was interpolating each quad into four smaller quads such that the new vertices were positioned based on the normals of each corner. For two adjacent corners, get the plane corresponding to each normal, with the plane touching a corner. Where the two planes intersect in a line, the new point is the closest point on that line to the existing edge between the two corners. The center point would be done similarly for diagonals, and the new point would be the average of the two points calculated that way. This should result in a quad subdivided so that the new points conform to whatever local curvature the object has.

adrianVmariano, 2025-05-08:
There's a larger question here than just quincunx.  Not sure if we can get Revar onboard to kill it...4 years later.  But we could deprecate it, presumably.    Though I note that the "hills" heightfield texture says it requires "quincunx" style in the docs.  

But what about all the other types?  Are any of them identical?  Can we provide any guidance about how to pick one more useful than "try them all"?