BelfrySCAD/BOSL2 #536: [BUG] setting a chamfer on prismoid() fails due to missing hull() function

dimo414, 2021-05-18:
```
use <BOSL2\std.scad>
prismoid(size1=[2,5], size2=[10,5], h=20, chamfer=1);
```

leads to:

```
Parsing design (AST generation)...
Saved backup file: C:/Users/Michael/Documents/OpenSCAD/backups/unsaved-backup-CVATgFGH.scad
Compiling design (CSG Tree generation)...
WARNING: Ignoring unknown function 'hull' in file ../../lib/BOSL2/shapes.scad, line 519 
ERROR: Assertion 'is_vnf(vnf)' failed in file attachments.scad, line 246 
TRACE: called by 'attach_geom' in file attachments.scad, line 959 
TRACE: called by 'attachable' in file vnf.scad, line 485 
TRACE: called by 'vnf_polyhedron' in file shapes.scad, line 465 
TRACE: called by 'children' in file attachments.scad, line 978 
TRACE: called by 'if' in file attachments.scad, line 977 
TRACE: called by 'if' in file attachments.scad, line 976 
TRACE: called by 'multmatrix' in file attachments.scad, line 968 
TRACE: called by 'attachable' in file shapes.scad, line 464 
TRACE: called by 'prismoid' in file laundry, line 3 
Compiling design (CSG Products generation)...
Geometries in cache: 76
Geometry cache size in bytes: 1144224
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized tree has 1 elements!
Compile and preview finished.
Total rendering time: 0:00:00.079
```

It looks like `include <BOSL2\hull.scad>` is all that's missing, but this shouldn't be an implicit requirement of using `prismoid()`.

Not sure if this is related, but it also seems like `prismoid()` doesn't support negative chamfers; setting `chamfer=-1` doesn't create negative chamfers, unlike `cuboid()` and `cyl()`.

revarbat, 2021-05-22:
The docs for `prismoid()` actually specifically state:

>Note: if using chamfers or rounding, you must also include the hull.scad file:
```openscadinclude <BOSL2/hull.scad>
```

However, I may have an idea to

revarbat, 2021-05-24:
Couldn't include hull.scad just on rounding/chamfer usage, so I moved hull.scad into std.scad.

closed 2021-05-24 by revarbat