BelfrySCAD/BOSL2 #549: [BUG] prismoid() doesn't support negative chamfer

dimo414, 2021-05-24:
**Describe the bug**
Setting a negative chamfer on `prismoid()` has no effect.

**Code To Reproduce Bug**
```
cmf=-1;
xdistribute(20) {
    cyl(r=5, h=10, chamfer=cmf);
    prismoid(size1=[2,5], size2=[10,5], h=20, chamfer=cmf);
}
```

**Expected behavior**
Negatives chamfers on both the cylinder and the prismoid.

**Screenshots**
* [`cmf=1`](https://user-images.githubusercontent.com/1885701/119321468-87a3a400-bc31-11eb-8f8a-66ccd10a6cec.png)
* [`cmf=-1`](https://user-images.githubusercontent.com/1885701/119321520-9722ed00-bc31-11eb-93ae-c33d7a8ca6b5.png)

**Additional context**
- BOSL2 Version: 2.0.638
- OpenSCAD Version: 2021.01

I mentioned this issue in #536 but I figured I should just file a separate issue.

adrianVmariano, 2021-05-28:
What does this even mean?  The rounding on prismoid is defined to work only for the vertical edges.  You can't do negative rounding on those edges because the direction of the rounding is ambiguous.  We went through this same issue with negative roundings on cubes, right?  

     rounded_prism(rect([10,40],center=true), rect([40,10],center=true), height=20, joint_top=-3, joint_bot=-9);

The above will do rounded top and bottom with negative rounding.   You can almost get chamfers by picking splinesteps=3.  Is that the type of shape desired?   It seems like a lot of trouble to add that to prismoid(), especially if you have rounded vertical edges.   (I'm actually not sure why I banned splinesteps<3.)

adrianVmariano, 2021-05-28:
See also issue #41

dimo414, 2021-05-28:
> The rounding on prismoid is defined to work only for the vertical edges. You can't do negative rounding on those edges because the direction of the rounding is ambiguous.

Ah interesting, I missed that distinction. That does make sense, I was confused because some of the modules in shapes.scad support negative chamfers/roundings and prismoid appeared to not. Since prismoid used to require hull.scad to get chamfers to work at all I erroneously assumed this was a bug related to hull-ing the shape.

> We went through this same issue with negative roundings on cubes, right?

Sorry, I don't know. I'm a new user. If negative roundings/chamfers are intentionally unsupported by certain shapes would it be possible to `assert()` that these values are not-negative?

> The above will do rounded top and bottom with negative rounding. You can almost get chamfers by picking splinesteps=3. Is that the type of shape desired?

Very cool, thank you for sharing! I will experiment with this further.

adrianVmariano, 2021-05-28:
cuboid actually accepts negative rounding for the top and bottom, just not the sides.  Prismoid needs an assert for negative rounding.  I think cuboid already asserts if you specify the vertical edges with negative rounding.  So if you give edges=FRONT with a negative radius it asserts.

adrianVmariano, 2021-05-29:
It occurs to me that the docs for prismoid should have a "see also: rounded_prism" and maybe mention that rounded_prism has additional rounding options.

revarbat, 2021-06-23:
Added asserts.

closed 2021-06-23 by revarbat