BelfrySCAD/BOSL2 #554: [BUG] rect_tube() makes only errors

3Dietrich, 2021-05-28:
**Describe the bug**
I tried today the rect_tube (https://github.com/revarbat/BOSL2/wiki/shapes.scad#module-rect_tube)
The examples 1..6 work, but from 7 on I got only errors. 
(and btw: the pictures in the Example 7++ show 'something' but it often looks not correct)

**Code To Reproduce Bug**
I just tried the given examples, like Example 7:
```
include <BOSL2/std.scad>
rect_tube(size=100, wall=5, rounding=10, irounding=0, h=30);
```
or e.g. Example 11:
```
include <BOSL2/std.scad>
rect_tube(
    size1=100, size2=80, wall=5, h=30,
    rounding1=10, rounding2=0,
    irounding1=8, irounding2=0
);
```
the log shows:
WARNING: Ignoring unknown function 'hull' in file ../../../../../OpenSCAD/libraries/BOSL2/shapes.scad, line 519 
ERROR: Assertion 'is_vnf(vnf)' failed in file ../../../../../OpenSCAD/libraries/BOSL2/attachments.scad, line 246 
TRACE: called by 'attach_geom' in file ../../../../../OpenSCAD/libraries/BOSL2/attachments.scad, line 959 
TRACE: called by 'attachable' in file ../../../../../OpenSCAD/libraries/BOSL2/vnf.scad, line 486 
TRACE: called by 'vnf_polyhedron' in file ../../../../../OpenSCAD/libraries/BOSL2/shapes.scad, line 465 
TRACE: called by 'children' in file ../../../../../OpenSCAD/libraries/BOSL2/attachments.scad, line 978 
TRACE: called by 'if' in file ../../../../../OpenSCAD/libraries/BOSL2/attachments.scad, line 977 
TRACE: called by 'if' in file ../../../../../OpenSCAD/libraries/BOSL2/attachments.scad, line 976 
TRACE: called by 'multmatrix' in file ../../../../../OpenSCAD/libraries/BOSL2/attachments.scad, line 968 
TRACE: called by 'attachable' in file ../../../../../OpenSCAD/libraries/BOSL2/shapes.scad, line 464 
TRACE: called by 'prismoid' in file ../../../../../OpenSCAD/libraries/BOSL2/shapes.scad, line 683 
TRACE: called by 'children' in file ../../../../../OpenSCAD/libraries/BOSL2/attachments.scad, line 1455 
TRACE: called by 'children' in file ../../../../../OpenSCAD/libraries/BOSL2/attachments.scad, line 1384 
TRACE: called by 'hide' in file ../../../../../OpenSCAD/libraries/BOSL2/attachments.scad, line 1455 

**Expected behavior**
well.. it should show the rect_tube. And less errors :)

**Screenshots**
there is nothing to show

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version: 2021.01
- Other libraries used: only BOSL2 (in this case)

RonaldoCMP, 2021-05-28:
To use shapes.scad you should include the hull.scad file as well as std.scad. The docs should mention that.

3Dietrich, 2021-05-28:
yes, it works. Thank you!
And yes: it should be added to each example code.

revarbat, 2021-05-29:
Update to the latest BOSL2 code.  The need for including hull.scad was removed a few days ao.

3Dietrich, 2021-05-29:
ah ok!
I just updated/downloaded BOSL2 on May 20 --> Already too old :)
After updating again: Yes, it works now without including hull.scad
ok, no bug.

But what about the pictures in the wiki (Exapmle 7,8,9,10,13,14,15), they look strange. e.g.
[image]

revarbat, 2021-05-29:
The automated docs generation system is run on a virtual machine spun up whenever a Pull Request is merged. I have yet to determine why it is sometimes generating bad images, when my own machine at home makes clean images.

closed 2021-05-30 by revarbat