BelfrySCAD/BOSL2 #609: New module to move children in an arbitrary direction

rjcarlson49, 2021-07-19:
I looked for functionality that would move an object in an arbitrary direction, but could not find it. Perhaps it there in a place I did not look. If not I suggest this addition.

```
// Lat is the angle of rotation around the X axis, with the Y-axis being 0 using the right hand rule
// Lon is the angle of rotation around the Z axis, with the X-axis being 0 using the right hand rule
// angelVector produces a unit vector oriented by the angles (lat, lon)
function angleVector(lat, lon) =
    [cos(lon) * cos(lat), sin(lon) * cos(lat), sin(lat)];

// move moves the children 'dist' units in the direction given by (lat, lon)
module move(lat, lon, dist) {
    translate(dist * angleVector(lat, lon)) children();
}
```

revarbat, 2021-07-19:
`move()` is already a module that exists. It's a shorter named equivalent of `translate()`. However, what you are asking for can be done with `move(spherical_to_xyz(r,theta,phi))`.

revarbat, 2021-07-19:
Where `theta` is degrees counterclockwise of X+ around the Z axis, and `phi` is degrees away from Z+.

revarbat, 2021-07-19:
Actually, there's also `move(altaz_to_xyz(alt,az,r))` which is a better fit.

rjcarlson49, 2021-07-20:
Thanks. You have a discoverability problem though. I could only find spherical_to_xyz when I already knew the answer. I suggest putting something in the Translations section. That's where I looked first. Movement in an arbitrary direction is pretty basic.

revarbat, 2021-07-20:
An example is indicated for `move()`

revarbat, 2021-07-20:
Also a candidate case for the See Also docs block.

closed 2021-07-23 by revarbat