BelfrySCAD/BOSL2 #620: Examples of 1/4 npt female / male threads and Schrader bike valve adapters

stillpointx, 2021-08-09:
Are there any examples of 1/4 NPT female / male threads and Schrader bike valve adapters?

I know I can create various threads using BOSL2 but how does one adapt the code to create 1/4 NPT female and Schrader bike valves?  Is there a example code / database for these threads?

Example:

include <BOSL2/std.scad>
include <BOSL2/threading.scad>
pitch = (2.16); //pitch Distance between threads. Default: 2mm/thread (2.28)

depth_a = pitch * cos(30)*5/8; // change 5/8 to adjust thread length

// d Inside base diameter of threads. Default: 10

thread_helix(d=39.34, starts=2, twist=190,profile = [
    [-5/16, 0           ],
    [-1/16, depth_a/pitch ],
    [ 1/16, depth_a/pitch ],
    [ 6/16, 0           ],
], higbee=3, $fn=72);

adrianVmariano, 2021-08-10:
Do you think there is some connection between 1/4 NPT threading and Schrader threading?  

There are examples of 1/2 inch NPT here:

https://github.com/revarbat/BOSL2/wiki/threading.scad#module-npt_threaded_rod

and it is very easy to change 1/2 to 1/4.  However, they are all male threading.  Probably there should be some examples of the internal female threads, though it's not hard: set internal to true and subtract from a cylinder, making sure you orient the taper the right way of course.  

I think thread_helix is the method of last resort.  You should be starting with one of the threaded rod options first.

stillpointx, 2021-08-10:
Thanks for the NPT information, I needed to download the new version of
BOSL2 to get it to work.  No I don't think 1/4 NPT is connected to Schrader
threading but I'm creating a 3D adapter to test the pressure of PLA /
PETG.  The pressure gauge is NPT and I'm goiing to use a Schrader connector
to pump air into to see where it fails.

-- 
-----
"There are two modes of being those in bliss and those seeking bliss,
may each step you take and each thought you make be made in a mindful and
blissful state."

This message is confidential and is intended only for the individual(s) or
entity named above and others who have been specifically authorized to
receive it. If you are not the intended recipient, please do not read,
copy, use or disclose the contents of this communication to others. Please
notify the sender that you have received this e-mail in error by replying
to the e-mail and then delete this email and any copies of it. Under no
circumstances may the contents hereof be disclosed to the public without
the sender's prior written permission.
-----

adrianVmariano, 2021-08-10:
https://www.ring-plug-thread-gages.com/PDChart/Tire-Valve-thread-data.html
Appears to have schrader specs.  The thread is vg8.  Presumably you should do OK if you use the average of the min and max diameters.  Note angle is 60 deg (what BOSL calls 30 deg) which would match the pattern of the triangular_threaded_rod, so maybe you can get that to make what you need.   These would be very demanding to print, I think, at 32 threads per inch.

stillpointx, 2021-08-11:
Thanks, I'll check it out.

-- 
-----
"There are two modes of being those in bliss and those seeking bliss,
may each step you take and each thought you make be made in a mindful and
blissful state."

This message is confidential and is intended only for the individual(s) or
entity named above and others who have been specifically authorized to
receive it. If you are not the intended recipient, please do not read,
copy, use or disclose the contents of this communication to others. Please
notify the sender that you have received this e-mail in error by replying
to the e-mail and then delete this email and any copies of it. Under no
circumstances may the contents hereof be disclosed to the public without
the sender's prior written permission.
-----

closed 2021-08-24 by adrianVmariano