BelfrySCAD/BOSL2 #624: Counter Intuitive or Bug - Translate in Y required to move in Z of attached child. [BUG]

123MT, 2021-08-11:
/*
Translate in Y required to move in Z of attached child.

Not sure if this is how the library is supposed to work.  You will see in the following example that in order to move the attached child cube in Z you need to add a translate to Y.  For me: Very. Confusing. Not. Intuitive.

I suspect this line in the tutorial might be a hint:
"The orientation of the child will be overridden to point outwards from the center of the parent, more or less:"

It would be nice to have a feature to turn off this override if this is what is happening.  It would be far more intuitive (to me) to imagine translating in space the child standalone prior to attachment instead of trying to determine where the x,y and z lie after attaching.  So I am attaching a ghost of the original position of the child to the parent, and the child is then being translated from that ghost using its original axis.  

OpenScad 2021.01, BOSL2 Library Downloaded: 7/11/2021
*/

include <BOSL2/std.scad>
if (1){
if (1) {
  cylinder(h=20, r=10, center=true)
  attach(FRONT,BACK)
  translate([0,10,0]) //need to translate Y to move in Z !?
  cube([10,5,20],center=true);
}else {
  translate([0,10,0]) //same translate moves Y as expected
  cube([10,5,20],center=true);
}}

//Expected that I would need to use this code to produce what above code produces:

else {
  cylinder(h=20, r=10, center=true)
  attach(FRONT,BACK)
  translate([0,0,10]) //<<<----NOTE CHANGE HERE.  THIS IS HOW I EXPECT TO CODE TO GET ABOVE RESULT
  cube([10,5,20],center=true);
}

adrianVmariano, 2021-08-12:
Actually I realized that the above example is a little potentially misleading.  If you put the translate outside (before the attach) then it acts in the parent coordinate system.  So:

    cylinder(h=20,r=10)
    translate([0,0,10])
    attach(FRONT,BACK)
    cube([10,5,20]);

will work as you desire.  If you use position() then you can have the translate inside:

    cylinder(h=20,r=10)
    position(FRONT)
    translate([0,0,10])
    cube([10,5,20],anchor=BACK);

revarbat, 2021-08-30:
Attachments tutorial is already being reworked to make this clearer.

revarbat, 2021-08-30:
This is being tracked in issue #618

closed 2021-08-30 by revarbat