BelfrySCAD/BOSL2 #628: import (svg) to path

3Dietrich, 2021-08-30:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Example Code**
```
// Code goes here.
```

**Additional context**
Add any other context or screenshots about the feature request here.

adrianVmariano, 2021-08-30:
Note that OpenSCAD doesn't have file reading commands.  You need to look for a program that will convert your SVG file into a point list, or maybe a bezier list, that you can then embed in your scad file.

revarbat, 2021-08-30:
Sadly, with OpenSCAD's current limitations, it isn't possible to import a path from an SVG.  I believe you *CAN* import an SVG as 2D geometry, but you can't get a path from the geometry shape.  OpenSCAD is silly like that.  Despite begging and pleading, it appears unlikely that this will ever change.

closed 2021-08-30 by revarbat

maxroseGIT, 2026-09-15:
@revarbat can you comment why this cannot be done in OpenSCAD? What is the silly thing that prevents it?

I don't understand, why it is possible to import a svg file that can be extruded, but not used otherwise as a path or similar?

In fact importing svg paths (or at least shapes, I don't know the correct term in scad slang for it) is possible:
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Importing_Geometry/SVG_Import

I think the limitation of svg import has been solved...

Can't we proceed from here?

adrianVmariano, 2026-09-15:
Nope.  OpenSCAD imports SVG as geometry, not as data.  There is no way to extract data (like point positions) from geometry.  There may eventually be a way to import an SVG as data, but not right now.  I guess if you convert your SVG to json with some external tool you could do it with the import() function.

maxroseGIT, 2026-09-16:
Thanks for feedback here.

May I ask what the difference between 'geometry' and 'data' is (in terms of openSCAD)? I mean geometry must also be some kind of data, or else it wouldn't be a digital thing? 

Btw. I found some tool doing the conversion of SVG path into points, but this is a tedious workaround (https://github.com/fablabnbg/inkscape-paths2openscad) to engage an export extension of inkscape to create a point list of certain precision, which then has to be included into an SCAD file.
This workaround, however, is quite a hindrance in an otherwise streamlined workflow in my opinion.

Anyone willing to give an explanation here, why geometry in openSCAD cannot be converted into point data inside of openSCAD?

revarbat, 2026-09-17:
Because, even though several attempts have been made to add said functionality, it has never been accepted into the codebase.

The last stable release of OpenSCAD was five years ago.

This is part of why there have been several different projects spawned by people being horribly frustrated by OpenSCAD and deciding to loudly roll their own. 

I started the BOSL2 library to fill in a bunch of the holes in OpenSCAD functionality. But there's only so much you can do in user-space. 

But, you, friend, may be lucky. You see, I also got horribly frustrated. I spent the last three months writing my own complete implementation of the OpenSCAD language and GUI, called BelfrySCAD. Faster. With debugging tools and visualizers.

BelfrySCAD has two ways to do the SVG path import:

1. `region = import("foo.svg");`

2. `obj = render() {import("foo.svg");};` then read the data from `obj.paths`, though you'll need to de-index the path points from the `obj.vertices`.

A region, BTW, is just a list of non-overlapping paths that represent perimeters and holes in an even-odd composition. See the BOSL2 region.scad docs.

https://github.com/BelfrySCAD/BelfrySCAD/

maxroseGIT, 2026-09-17:
@revarbat thanks for this very positive answer in terms of a solution to my question and even the OP.

I stumbled over this preference setting of dev 2026.08.31 (git c55e8eff) version, that actually shows this option under the features tab:

- Enable import function retruning data instead of geometry

[image]

I wonder how this is used... it is not documented in any obvious way...

adrianVmariano, 2026-09-17:
The import() function in OpenSCAD only imports json files and the developers have no current plans to add svg support.

maxroseGIT, 2026-09-17:
Hi @adrianVmariano, 
thanks for comment on that.

What exactly do you mean by no current plans to add svg support?

In fact the import function can import svg (at least some meaningful geometry part of it)
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Importing_Geometry/SVG_Import

Do you mean that import does not interpret any svg commands?

adrianVmariano, 2026-09-17:
Geometry and data are different.  Functions and modules are different.  The import MODULE is what you referred to.  It imports SVG into geometry---the picture you can see on the screen.  Which cannot be turned into data.  The import function reads only json files.  It does not interpret svg commands because that has nothing to do with json parsing.  They just fixed it to give you an error if you try to read a .svg file instead of stupidly trying to interpret the svg data as json.