BelfrySCAD/BOSL2 #631: Can we improve doc front page, and have better file summaries?

adrianVmariano, 2021-09-04:
In an email exchange it was suggested to me that the functions of the various BOSL2 files is obscure.  The person I was emailing suggested that we should have longer, more detailed descriptions on the front page of what each file does. And at the top of each file, much longer descriptions of the contents of that file.   Who is this file for?   (e.g.  WTF are quaternions?)  

I also wonder about the right order for things.  I've seen it happen in individual files that things seem like they are in the wrong order, an order that makes sense to the programmer but not to the reader.   For example, having generic threading appear before standard threading doesn't really make sense to the reader, who most likely cares about standard threading.   So what's the right order to present things on the front page?   (And similarly, do any individual files need to be reordered so that the presentation in the docs makes more sense to a user?)  

The person I was emailing also singled out "mutators" as a mystifying term, whose meaning was totally unclear.  Can we find a better term?  It's sure not the place I would look for "rainbow".

revarbat, 2022-01-05:
This is actually four issues:
[x] Docsgen needs File Summaries, Groups, and ordering.  This has been implemented.
[ ] Ordering of docs sections within each file.  This is ongoing, but @adrianVmariano has already fixed much of this.
[ ] A better name for mutators.scad.  Well, we haven't really come up with a better name yet.
[ ] Should colors code `HSV()`, `HSL()`, `rainbow()` be moved to a colors.scad?

adrianVmariano, 2022-01-05:
I think that maybe the contents of mutators need to be dispersed to other files.   We put the half-space stuff into partitions, for example.   It makes sense to put the color stuff in colors.scad, though not sure having such a tiny file makes sense.  I don't know---does it matter?  Certainly from a docs perspective, having color stuff in its own file would be better.  (Is there anything else that needs doing for color support?)

There are about 5 files left that need serious evaluation from the core files.  I was not planning to review all the parts files.

adrianVmariano, 2022-03-26:
Should recolor and color_this be moved to colors.scad?

revarbat, 2022-03-27:
Thematically, yes.

closed 2022-04-22 by adrianVmariano