BelfrySCAD/BOSL2 #633: "assemble_a_path_from_fragments()" doesn't work or needs an example

3Dietrich, 2021-09-05:
**Describe the bug**
I tried to combine two pathes into one using [assemble_a_path_from_fragments()](https://github.com/revarbat/BOSL2/wiki/paths.scad#function-assemble_a_path_from_fragments) or [assemble_path_fragments()](https://github.com/revarbat/BOSL2/wiki/paths.scad#function-assemble_path_fragments). Although the description sounds right for my usage, both didn't work, neither using extra '[]' or not. 

This is probably my mistake, I just don't know how to find the error. Already "concat" - is not to be found..
Therefore the examples are so important..

**Code To Reproduce Bug**
```
    // simplified form, normally an svg form, converted to [x,y]
    path = [ [0,0], [10,0] , [10,10], [10,0] ];
    //get the mirror of 'path' and put them together as one
    fulpath = assemble_a_path_from_fragments( [path, mirror([1,0], p=path)] );
    /*  repalcing 'fulpath' with the following doesn't work either:
       = assemble_a_path_from_fragments( path, mirror([1,0], p=path) );
              = assemble_path_fragments([path, mirror([1,0], p=path)]);
              = assemble_path_fragments( path, mirror([1,0], p=path) );
    */
    // making a 3d object out of the fulpath and fulpath with offset
    skin( [offset(fulpath, r=2, closed=true), offset(fulpath, r=11, closed=true)], z=[0,22], slices=2);
```

**Expected behavior**
there should be less blue and red error messages but a visible 3d object.
I guess that is need to avoid the "[]" between the two pathes? But I couldn't find the function (

Assertions like: 'is_path(poly, dim = 2)' failed: "Input should be a 2d path" in file ../../../..
doesn't help me, because it *is* 2d path..

**Additional context**
- OpenSCAD Version: 2021.01
- Other libraries used: [2, 0, 641]

adrianVmariano, 2021-09-05:
Indeed if you echo(fulpath) you can see it is a list of paths instead of a path, so no, it is *not* a 2d path.    I think assemble_a_path_from_fragments is a specialized function in support of boolean path operations.  The documentation for it is terrible, so I'm not sure what it does, but I suspect it's not what you need.

Can you explain what you're trying to do?  It sounds like you probably want something like `concat(path,reverse(xflip(path))`, but I'm not sure.  Also your path backtracks on itself, which will not work in skin().    Maybe you meant the last point to be [0,10]?

adrianVmariano, 2021-09-06:
Actually also you probably need to run deduplicate if you do the above because offset will be unhappy with the repeated points on the Y axis.

RonaldoCMP, 2021-09-06:
Your `path`, the first, folds on itself; the second path, a mirror of it,
does not close the first path.

The error message comes from skin; it rejects your `fulpath` because it is
not a 2d path. Echo it to see.

The function `assemble_a_path_from_fragments` always returns a pair where
the former element is the possibly assembled path and the later a list of
unused fragments. You should input `skin` with just the former element,
which is a 2d path.

3Dietrich, 2021-09-06:
> Maybe you meant the last point to be [0,10]?
yes! Damn, I wanted to make it simple and easy but now I added a stupid error. The path should be
`    path = [ [0,0], [10,0] , [10,10], [0,10] ];
`

3Dietrich, 2021-09-06:
> Can you explain what you're trying to do?

It should be a symmetric form, so I constructed one half of the object ('path') and want to combine it with the mirror at x-axis of itself, what should result the complete 2d form ('fulpath'). Then, with skin, I want to make it 3D using fulpath (with some resizing 'offset') on different z hights.

Here is a screenshot of what I want ( here I used hull() with 2x linear_extrution(1) ):
[image]
the red area is the 'path'

3Dietrich, 2021-09-06:
Thank you @adrianVmariano, your reaction was very helpful and the right advices.
fyi, this is working now:
```
    path = [ [0,0], [10,2] , [10,5], [0,7] ];
    fulpath = deduplicate(closed=true, concat( path,reverse(xflip(path))));
    skin( [offset(fulpath, r=2, closed=true), offset(fulpath, r=11, closed=true)], z=[0,22], slices=2);
```
to get all the details known is a pain (e.g. knowing "deduplicate"! and knowing that adding "closed=true" is fully important or you get red and blue errors instead of 3D object) but this is not you but my problem.. 
And at least I can sometimes imagine something out of the error messages.
Thank you again!
<close>

closed 2021-09-06 by 3Dietrich

adrianVmariano, 2021-09-06:
What I wrote already is the solution to your problem.  The simplest solution for your example would be:

      $fn=36;
      path = [[10,0],[8,10]];
      fulpath = concat(path, reverse(xflip(path)));
      skin( [offset(fulpath, r=2, closed=true), offset(fulpath, r=11, closed=true)], z=[0,22], slices=2, method="tangent");

Note that there was a---call it a bug---in offset() where it didn't respect $fn, $fa and $fs properly that I fixed last week.  (There is an undocumented parameter called maxstep that controls the segment length.)   This makes it harder to get the profiles to line up for skinning.    If you update to the latest version you can use $fn to control the number of points at the corners and it comes out the same at each corner, so the profiles line up neatly for skinning.  Then you can skip the "tangent" method.  Otherwise, you need the tangent method to get a proper alignment and it's kind of slow because offset makes a lot of points, a lot more than you need.  (In the old BOSL2 version you can set maxstep=0.5 in the offset calls to get a faster result.)  

Depending on what you want to do, offset_sweep() may be worth a look because it knows where the points produced by offset came from so it can produce perfect point alignments from layer to layer.  

Note that in my design above, I omit the points on the Y axis because if you include them then they get duplicated when you mirror them, and that is an extra complication: offset will fail with a very cryptic error message.  (I fixed that.)  If you have duplicate points you need to remove them with deduplicate().  

Ronaldo, error comes from offset, not skin.  

And the "bug" here is that the path fragments functions need to be documented or maybe just hidden from users (made into internal functions).