BelfrySCAD/BOSL2 #700: Easy named attachments

ssendev, 2021-10-17:
**Is your feature request related to a problem? Please describe.**
While writing an object that should attach to another object it would be nice to reuse existing attachment points of a child object.

**Describe the solution you'd like**
In the example it would be nice if `body_with_handle` could be generated without using math.

Ideally it would be possible to replace `anchor_arrow` with `add_attachment("hand")` but i guess that's not possible with OpenScads computation model.

Maybe it's possible to a least help with the generation of `body_with_handle` by printing out the coordinates of the anchor point which could then be pasted. Usage would probably look similar to `expose_anchors` so `print_anchors() { body() print_anchor(); }`

It would also be nice to have an option to preserve the original anchor points when adding named ones.

**Describe alternatives you've considered**
Calculating the anchors by hand which at least for me means guessing numbers till it looks good enough;

**Example Code**
```OpenScad

module toy() {
  cyl(d=0.5,h=15, anchor=BOTTOM) {
    position(TOP) cuboid([5,0.5,3], anchor=LEFT+TOP);
  }
}
module pole() {
  cyl(d=1, h=30, anchor=BOTTOM) {
    children();
  }
}
module body() {
  cyl(d1=10, d2=5, h=12, anchor=BOTTOM) {
    attach(TOP) sphere(d=10); // head
    attach(LEFT) cyl(d=3, h=5, anchor=BOTTOM) {
      attach(TOP) sphere(d=4, spin=0) { // left hand
        children();
      }
    }
    attach(RIGHT) cyl(d=3, h=5, anchor=BOTTOM) {
      attach(TOP) sphere(d=4) { // right hand
        attach(TOP+FWD) anchor_arrow();
      }
    }
  }
}
module body_with_handle(anchor=CENTER, spin=0, orient=UP) {
  anchors = [
      named_anchor("hand", [10, 0, 6], [1,0,-0.6], 0)
      // would be nice to derive these numbers based on anchor_arrow/print_anchor
  ];
  attachable(anchor, spin, orient, r1=10, r2=8, l=15, anchors=anchors) {
    // guessed some numbers to restore the behavior of body()
    // (yes this is completely broken since it would have to be a small sphre that's offset
    // from the origin, but for the one used orientation looks more or less ok)
    body();
    children();
  }
}

left(30) body() {
  attach(TOP+BACK) toy(); // putting toy in hand is easy
}

pole() {
  attach(TOP, "hand")  body_with_handle() { // putting body on pole is hard
    attach(TOP+LEFT) toy(); // now putting toy in hand is also hard
  }
}
```

**Additional context**
[image]

revarbat, 2021-11-13:
This is how I'd code this up.  Calculations for left and right hands are symmetric in the X direction. The calculations are based on the height and cone angles of the body and arms.

```openscad
module toy() {
  cyl(d=0.5,h=15, anchor=BOTTOM) {
    position(TOP) cuboid([5,0.5,3], anchor=LEFT+TOP);
  }
}

module pole() {
  cyl(d=1, h=30, anchor=BOTTOM) {
    children();
  }
}

module body(anchor=CENTER, spin=0, orient=UP) {
  mat = (
      up(6) *  // Half height of body.
      rot(from=UP,to=[12,0,5/2])  // Angle of body cone side.
    ) * (
      up(5)  // Height of arm cylinder.
    );
  pos = apply(mat,[0,0,0]);
  vec = unit(apply(mat,UP) - pos, UP);
  anchors = [
      named_anchor("lhand", pos, vec, 0),
      named_anchor("rhand", v_mul(pos,[-1,1,1]), v_mul(vec,[-1,1,1]), 0)
  ];
  attachable(anchor, spin, orient, r1=10, r2=8, l=15, anchors=anchors) {
    cyl(d1=10, d2=5, h=12, anchor=BOTTOM) {
      attach(TOP) sphere(d=10); // head
      attach(LEFT) cyl(d=3, h=5, anchor=BOTTOM) {
        attach(TOP) sphere(d=4, spin=0); // left hand
      }
      attach(RIGHT) cyl(d=3, h=5, anchor=BOTTOM) {
        attach(TOP) sphere(d=4) { // right hand
          attach(TOP+FWD);
        }
      }
    }
    children();
  }
}

left(30) body() {
  attach("rhand") toy();
  attach("lhand") toy();
}

pole() {
  attach(TOP, "rhand") body() {
    attach("lhand") toy();
  }
}
```

revarbat, 2022-01-05:
I'm afraid the original request isn't reasonably implementable in OpenSCAD.

closed 2022-01-05 by revarbat